ceccocats/tkDNN #110: how can i use libtkDNN.so in a new project? please help

jcyhcs, 2020-08-17:
ok, i use have a new c++11 project, use cmake as compile. i install tkDNN in /usr/local/lib, and tkDNN include as /usr/local/include/tkDNN,  but when i compile i get error:
undefined reference to symbol 'cudaGetErrorString@@libcudart.so.10.0'

how can i use libtkDNN.so into my project?? 
my cmakefile.txt like this:
[image]

ceccocats, 2020-08-17:
You need to link all the libraries
The easyest way is to use `find_package(tkDNN)`
It will load the config from:
https://github.com/ceccocats/tkDNN/blob/master/cmake/tkDNNConfig.cmake

You will have the include dirs in `tkDNN_INCLUDE_DIRS` and the libraries to link in `tkDNN_LIBRARIES`

nightduck, 2020-11-19:
Stumbled on this issue, figured out the specifics. For future googlers after me, you have to install this repo, by running `sudo make install` after building. Then link it by adding the relevant code below to your CMakeLists.txt file:

    find_package(tkDNN REQUIRED)
    
    include_directories(
        ${tkDNN_INCLUDE_DIRS}
        /usr/local/include/tkDNN
    )
    
    add_executable(program_name src/program_code.cpp)
    target_link_libraries(program_name ${tkDNN_LIBRARIES})

If @jcyhcs has fixed his problem, you can probably close this issue

mive93, 2020-11-19:
Yes, @nightduck that is correct.
If you want another real example where I use it as submobule you can check this https://github.com/mive93/class-edge/blob/master/CMakeLists.txt

I will close the issue for now, feel free to reopen it.

closed 2020-11-19 by mive93