ceccocats/tkDNN #111: FPS low with save_result = true

Mattis95, 2020-08-18:
I'm running a custom trained yolov4 tiny on the Jetson Nano. For inference I basically changed the demo file, so the input is the camera connected to the Jetson. Thats working fine and I get the usual ~30FPS. 
When I set the SAVE_RESULT variable on true to save the Video, the FPS is mutch lower, around 10-15 FPS.
Is this a common problem or am I doing something wrong?

ceccocats, 2020-08-19:
Its normal, SAVE_RESULT save the video with the detection so it takes some time to encode and write it.

closed 2020-08-19 by Mattis95

ahmettarhan, 2020-09-11:
Hi Mattis95,

how did you change the line ../demo/yolo_test.mp4  to use camera input instead of video file and what kind of camera did you connect to the jetson?

thanks in advance

Mattis95, 2020-10-10:
Sorry, I just seen it now. I used a USB camera, basically any USB camera should work. The Raspberry camera v2 also works.
If you want to use the camera, you just write 0 instead of the path to the video, then the camera input is used.