ceccocats/tkDNN #112: For fp32 and fp16 model,have the same FPS

basailuonaaa, 2020-08-19:


ceccocats, 2020-08-19:
Refer to this #38 
Your GPU is probably not powerful enough

basailuonaaa, 2020-08-19:
Thanks for your reply. I guess that's not the reason,I'm using nvidia RTX2080Ti

ceccocats, 2020-08-19:
Can you explain better the issue? What model are you testing? What commands have you done?

basailuonaaa, 2020-08-20:
Hi,@ceccocats, Thanks for your reply. 
Here my test script:
#include <iostream>
#include <signal.h>
#include <stdlib.h>
#include <unistd.h>
#include <mutex>
#include <time.h>
#include <sys/timeb.h>
#include "Yolo3Detection.h"
#include <chrono>

using namespace std::chrono;
#define TIMEB timeb

int main(int argc, char *argv[]){
    struct TIMEB ts1, ts2;
    time_t t_sec,ti;
    // Network name
    std::string net = "/home/wangcq/yolo_v4/source_code/TensorRT/tkDNN-darknet-demo/rt-model/batch==4/model_fp32.rt";
    if(argc > 1)
        net = argv[1];

    // Inference batch size
    int n_batch = 4;
    if(argc > 2)
        n_batch = atoi(argv[2]);

    // Number of classes
    int n_classes = 1;
    if(argc > 3)
        n_classes = atoi(argv[3]);

    tk::dnn::Yolo3Detection yolo;

    tk::dnn::DetectionNN *detNN;
    detNN = &yolo;
    detNN->init(net, n_classes, n_batch);

    cv::Mat img;
    std::vector<cv::Mat> batch_frame;
    std::vector<cv::Mat> batch_dnn_input;
    auto start = high_resolution_clock::now();
    // cv::namedWindow("detection", cv::WINDOW_NORMAL);

    // Read all images from a directory
    cv::String path("/home/wangcq/MyImages/helmet/test/*.*");
    std::vector<cv::String> fn;
    cv::glob(path,fn,true); // recurse
    int total = 0;
    int num_img = 0;
    std::cout<<"detection start\n";
    ftime(&ts1);//开始计时
    for (size_t k=0; k < fn.size(); ++k){
        img = cv::imread(fn[k], cv::IMREAD_COLOR);
        if (!img.data){
            std::cout << "Problem loading image!";
            break;
        }

        // this will be resized to the net format
        batch_dnn_input.push_back(img.clone());
        ++num_img;

        ++total;
        // do inference
        if (num_img == n_batch){
            detNN->update(batch_dnn_input, n_batch);
            num_img = 0;
            batch_dnn_input.clear();
        }
    }

    ftime(&ts2);//结束计时
    t_sec=ts2.time-ts1.time;//计算秒间隔
    double t_ms=ts2.millitm-ts1.millitm;//计算毫秒间隔
    ti=t_sec*1000+t_ms;
    std::cout<<"检测总耗时: time ="<<ti<<std::endl;
    std::cout<<"检测平均耗时: time ="<<ti/total<<std::endl;

    std::cout<<"detection end\n";

    auto stop = high_resolution_clock::now();
    auto duration = duration_cast<seconds>(stop - start);
    double mean = 0;
    std::cout<<COL_GREENB<<"\n\nTime stats:\n";
    std::cout << "Total files: " << total << " files\n";
    std::cout << "Total inference time: " << duration.count() << "s\n";
    std::cout<<"Min: "<<*std::min_element(detNN->stats.begin(), detNN->stats.end())/n_batch<<" ms\n";
    std::cout<<"Max: "<<*std::max_element(detNN->stats.begin(), detNN->stats.end())/n_batch<<" ms\n";
    for(int i=0; i<detNN->stats.size(); i++) mean += detNN->stats[i]; mean /= detNN->stats.size();
    std::cout<<"Avg: "<<mean/n_batch<<" ms\t"<<1000/(mean/n_batch)<<" FPS\n"<<COL_END;

}

I have run the code on a nvidia RTX2080Ti(ubuntu16.04, TensorRT-6.0.1.5, cuda-10.0, cudnn-7.6.1), on 1514 mages and this is what I get:
For YoloV4 608*608 model:

for FP32.rt:
Batch = 1 Avg: 15.0531 ms 66.4315 FPS
In addition, the time of the entire for loop is 42702ms,the average time to process each picture is 28ms;
Batch = 4 Avg: 13.3748 ms 74.7676FPS
In addition, the time of the entire for loop is 40573 ms,the average time to process each picture is 26ms;

for FP16.rt:
Batch = 1 Avg: 8.89959 ms 112.365 FPS
In addition, the time of the entire for loop is 33509 ms, the average time to process each picture is 22 ms;
Batch = 4 Avg: 6.85899 ms 145.794 FPS
In addition, the time of the entire for loop is 31408 ms,the average time to process each picture is 20ms;

for int8.rt:
Batch = 1 Avg: 7.38597 ms 135.392 FPS
In addition, the time of the entire for loop is 30809 ms,the average time to process each picture is 20ms;
Batch = 4 Avg: 5.97115 ms 167.472 FPS
In addition, the time of the entire for loop is 31217 ms,the average time to process each picture is 20ms;

The fps is gradually increasing, but the average time to process each picture is basically the same. Is this a normal phenomenon?

mive93, 2020-09-11:
Hi @basailuonaaa 

This is what I get on the same exact GPU
```
FP32, B=1
Avg:		16.5419 ms	60.4527 FPS
Avg full loop:	21.8497 ms	45.7673 FPS

FP32, B=4
Avg:		12.4958 ms	80.0272 FPS
Avg full loop:	16.5318 ms	60.4895 FPS
```

So yes, you're values are correct, but why? 
- the demo is just a demo, it's not the most optimized way to read input, that can be improved
- the preprocessing and postprocessing are executed on CPU and are sequential, even when using batches. Yes, that could be otpimized. 
- indeed what tKDNN wants and do optimize at best is the pure inference.

Let me know if you have further questions.

closed 2021-01-04 by mive93