ceccocats/tkDNN #113: Bouding boxes coords

rafoliveira, 2020-08-19:
Having a hard time understanding the code since I am not very experienced with C++
Which variable describes the coordinates of the bounding boxes?
I wanted to use them to make calculations

rafoliveira, 2020-08-26:
Found it on this [file](https://github.com/ceccocats/tkDNN/blob/master/src/Yolo3Detection.cpp), described by the _res_ variable

closed 2020-08-26 by rafoliveira

mive93, 2020-08-26:
Hi @rafoliveira 
Sorry, I missed the issue. 
The boxes are stored as tk::dnn::box (defined here https://github.com/ceccocats/tkDNN/blob/master/include/tkDNN/Layer.h#L577 )
```
struct box {
    int cl;
    float x, y, w, h;
    float prob;
    std::vector<float> probs;

    void print() 
    {
        std::cout<<"x: "<<x<<"\ty: "<<y<<"\tw: "<<w<<"\th: "<<h<<"\tcl: "<<cl<<"\tprob: "<<prob<<std::endl;
    }
};
```
For each detector, they are stored in batchDetected (https://github.com/ceccocats/tkDNN/blob/master/include/tkDNN/DetectionNN.h#L71)

sharoseali, 2021-01-11:
HI! I tested both  FP32 and FP16 implementation with batch size = 4 but when I run the test I get this result, Can anyone help. I think it's an NMS problem but don't know how to fix it.
[image]

sharoseali, 2021-01-15:
> HI! I tested both FP32 and FP16 implementation with batch size = 4 but when I run the test I get this result as seen below. Every object is surrounded by 4 bounding boxes., Can anyone help? I think it's an NMS problem but don't know how to fix it.
> [image]
> Thank You

@mive93

mive93, 2021-01-22:
Hi @sharoseali 

what are you using? Which platform?
I never had that problem. 
Just run a bunch of models and everything is fine for me.

sharoseali, 2021-01-22:
> Hi @sharoseali
> 
> what are you using? Which platform?
> I never had that problem.
> Just run a bunch of models and everything is fine for me.

İ am running my custom trained yolo v4  on AGX xavier using your repo. When İ use batch size equals 1 .. its fine, but when i use batch size equals 4 then on display i see 4 bounding boxes running with each moving object as İ show above. What the thing come in mind is that its issue in NMS.. and bounding boxes are 4 times drawn .Also FPS get very less when i run with batch size equals 4. thanks for replying