ceccocats/tkDNN #114: The problem when I use tkDNN in a new cmake project

shallowdream-x, 2020-08-20:
I run the cmake .. && make && sudo make install 
After that, include files are in /usr/local/include/tkDNN, lib files are in /usr/local/lib, the cmake files are in /usr/local/share/tkDNN

My new cmake project CmakeList.txt is:
SET(tkDNN_DIR /usr/local/share/tkDNN/cmake)

#tkDNN
find_package(tkDNN REQUIRED)
message(STATUS "tkDNN library status:")
message(STATUS "    libraries: ${tkDNN_LIBRARIES}")
message(STATUS "    include path: ${tkDNN_INCLUDE_DIRS}")

add_executable(test_tkDNN src/test/test_tkDNN.cpp)
target_link_libraries(test_tkDNN ${tkDNN_LIBRARIES})

when I run the cmake .. : the output is :
-- Found tkDNN
-- Found CUDNN: /usr/lib/x86_64-linux-gnu/libcudnn.so
-- Found CUDNN include: /usr/include
-- Found NVINFER: /usr/lib/x86_64-linux-gnu/libnvinfer.so
-- Found NVINFER include: /usr/include/x86_64-linux-gnu
-- tkDNN library status:
--     libraries: tkDNN;kernels;/usr/local/cuda-10.2/lib64/libcudart_static.a;Threads::Threads;dl;/usr/lib/x86_64-linux-gnu/librt.so;/usr/lib/x86_64-linux-gnu/libcublas.so;/usr/lib/x86_64-linux-gnu/libcudnn.so;/usr/lib/x86_64-linux-gnu/libnvinfer.so;opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab
--     include path: /usr/local/cuda-10.2/include;/usr/include;/usr/include/x86_64-linux-gnu

but when I run make, I encounter an error:
In file included from /home/jingxin/try/dancer-vision/src/test/test_tkDNN.cpp:11:0:
/usr/local/include/tkDNN/test.h:2:10: fatal error: tkdnn.h: No such file or directory
 #include <tkdnn.h>
          ^~~~~~~~~
compilation terminated.
I don't know how to solve this, can you help me!!!

ceccocats, 2020-08-24:
Add
```
include_directories(${tkDNN_INCLUDE_DIRS})
```
After finding tkDNN

omaralvarez, 2020-08-28:
Right now I don't think the library generation is properly done in modern CMake. I think I will create a PR if you guys think it would be cool?

closed 2021-11-25 by mive93