ceccocats/tkDNN #115: activation not supported: leaky

Rhee-site, 2020-08-21:
Hi, I have a problem.

Model provided in "Existing tests and supported networks" operate normally.
But the customized model does not work.
The customized model is modified from the csresnext50-panet-spp model.

The method using the customized model is as follows.

1. Clone the repo darknet follow the instructions and generate two directories debug and layers (they contain weights)

2. Make a copy of the csresnext50-panet-spp.cpp(Test.cpp) in tkDNN/tests/darknet/

3. Create a directory for my model in the build folder
And dump layers and debug generated from 1.
Also, add Test.cfg, Test.names in /tkDNN/tests/darknet/cfg, /tkDNN/tests/darknet/names

4. run from the build
$cmake ..
$make

5. ./test_Test

But Result is **activation not supported: leaky**

In the introduction, the leaky function among the activation functions is indicated as being provided, and as a result of checking DarknetParser.cpp, it was confirmed that it is specified.

please solution.

mive93, 2020-08-26:
Hi @Rhee-site 
yes, leaky is definitely supported by tkDNN. 

1. Did everything worked fine? Do you have both wieghts and outputs?
2. ok, did you changed names and paths inside? 
3. Perfect

Can you share the custon cfg?

mive93, 2021-01-04:
Closing for now, feel free to reopen.

closed 2021-01-04 by mive93

dylanstouls-bib, 2021-03-01:
Hi,

I'm a bit late, but I hope my answer will help in the future.

I had the same issue after creating a custom .cfg file on a Windows machine.
As you can see here : https://manpages.ubuntu.com/manpages/bionic/en/man1/dos2unix.1.html a text file made on a Windows machine could be different from a text file made on a Unix machine.

So, using `$ dos2unix myconfigfile.cfg` resolved my issue.

yotamraz, 2022-04-27:
@ghost solution did not work for me.
still getting `activation not supported: leaky` on the first layer.

**Running with:**
Jetson xavier NX
tkDNN v0.5
cuDNN v8.201
FP16 enabled
yolo4tiny