ceccocats/tkDNN #119: tkdnn error while loading shared libraries: libkernels.so

zhanghua7099, 2020-09-08:
Hi!
I have compiled this project and install it on my computer. I used the following orders:

```
git clone https://github.com/ceccocats/tkDNN
cd tkDNN
mkdir build
cd build
cmake .. 
make
sudo make install
```

The demo in tkDNN/build runs well. But when I try to use tkDNN in another project, I can't use it. I have added the following code in `CMakeLists.txt`:
```
find_package (tkDNN)
include_directories( ${tkDNN_INCLUDE_DIRS})
target_link_libraries( ${tkDNN_LIBRARIES})
```

The new project can be compiled, but it will show this error message:
`error while loading shared libraries: libkernels.so: cannot open shared object file: No such file or directory
`
It seems that the new project can't compile `libkernels.so`, but it is existed in `tkDNN/build`. Can you help me deal with this problem? Thank you!

Dadamsmeade, 2020-09-11:
Hi @zhanghua7099,

I also ran into this same issue, I'm somewhat of a beginner when it comes to cmake and library linking. However, I discovered that /usr/local/lib was not included in my LD_LIBRARY_PATH environment variable so I ran the following command:

export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:/usr/local/lib"

I also added /usr/local/lib to my /etc/ld.so.conf file.

This fixed things for me, hope it helps!

mive93, 2020-09-11:
Hi @zhanghua7099,

check also #77 and #114.
Let us know if you solved the problem.

closed 2021-01-04 by mive93