ceccocats/tkDNN #120: data copy back on CPU

qqcce, 2020-09-09:
@omaralvarez @ceccocats @FisherTiger95 @mive93 @AlessioLei94 
When I was testing yolov3, I found preprocess slow in Yolo3Detection.cpp
line 81 and 82

`bgr[ch].download(bgr_h); //TODO: don't copy back on CPU `
`checkCuda( cudaMemcpy(input_d + i*size + netRT->input_dim.tot()*bi, (float*)bgr_h.data, size*sizeof(dnnType), cudaMemcpyHostToDevice));`

when i try to copy from GPU to GPU, 
`checkCuda( cudaMemcpy(input_d + i*size + netRT->input_dim.tot()*bi, (float*)bgr[ch].data, size*sizeof(dnnType), cudaMemcpyDeviceToDevice));`

the result is totally wrong.

Copy back on CPU seems to be redundant but it somehow changed the result, can anyone tell me why, how to fix it?

omaralvarez, 2020-09-09:
You need to prepare the batch correctly. You should perform several `cudaMemcpy`, one for each channel. TensorRT consumes images in planar fashion:

https://docs.nvidia.com/deeplearning/tensorrt/developer-guide/index.html#data-format-desc

qqcce, 2020-09-10:
> You need to prepare the batch correctly. You should perform several `cudaMemcpy`, one for each channel. TensorRT consumes images in planar fashion:
> 
> https://docs.nvidia.com/deeplearning/tensorrt/developer-guide/index.html#data-format-desc
> 
> If you want it to be fast, you should use `libnpp`.

Hi @omaralvarez 
Thanks for your quick reply.
I copyed each channel with following code, still no detection, is there anything I miss?

original code:
```
for(int i=0; i<netRT->input_dim.c; i++) {
    int size = imagePreproc.rows * imagePreproc.cols;
    int ch = netRT->input_dim.c-1 -i;
    bgr[ch].download(bgr_h); //TODO: don't copy back on CPU
    checkCuda( cudaMemcpy(input_d + i*size + netRT->input_dim.tot()*bi, (float*)bgr_h.data, size*sizeof(dnnType), cudaMemcpyHostToDevice));
}
```

my code:
```
for(int i=0; i<netRT->input_dim.c; i++) {
    int size = imagePreproc.rows * imagePreproc.cols;
    int ch = netRT->input_dim.c-1 -i;
    checkCuda( cudaMemcpy(input_d + i*size + netRT->input_dim.tot()*bi, (float*)bgr[ch].data, size*sizeof(dnnType), cudaMemcpyDeviceToDevice));
}
```

I am trying to optimize the speed, can you give more information about libnpp?

mive93, 2020-09-11:
Thank you @omaralvarez for the quick reply. 
However, I don't think the answer solves the problem. 
Yes the order of the channel matters, but I also tried the code with the correct order and I had the same problem. Without the download I was not able to get the correct output. That's why I kept that annoying download. 
I have tried to solve it in several ways, but still I couldn't find a correct solution. But actually I have never tried libnpp, don't know if that could work.

omaralvarez, 2020-09-11:
I think it may have to do with this:

Note
In contrast with Mat, in most cases GpuMat::isContinuous() == false . This means that rows are aligned to a size depending on the hardware. Single-row GpuMat is always a continuous matrix.
You are not recommended to leave static or global GpuMat variables allocated, that is, to rely on its destructor. The destruction order of such variables and CUDA context is undefined. GPU memory release function returns error if the CUDA context has been destroyed before.

Maybe creating your own cuda buffer and then creating the Mat object with your dimension/stride/type info will be the solution.

qqcce, 2020-09-13:
> I think it may have to do with this:
>
> Note
> In contrast with Mat, in most cases GpuMat::isContinuous() == false . This means that rows are aligned to a size depending on the hardware. Single-row [...]

Thanks for your instruction, I will try that.

closed 2021-02-15 by qqcce

LouisDLK, 2021-07-05:
Hi @qqcce ,

Did you acheive to remove GpuMat to Mat download?
As you, I get no detection after cpu download removal.

Thanks,
Louis