ceccocats/tkDNN #122: Connecting IP Camera

ahmettarhan, 2020-09-11:
@ceccocats  @mive93  I have an ip camera connected to the jetson nano and i can stream it with video player. But how can i change the line ../demo/yolo_test.mp4 so that it processes ip camera input instead of video file?

mive93, 2020-09-11:
Hi @ahmettarhan,

check #96. 
If you have further question, just ask.

ahmettarhan, 2020-09-11:
Thank you for your quick response mive93,

I already checked #96 but I am getting GStreamer warning: Error opening bin: no source element for URI "dev/video0". And when run  ls -ltrh /dev/video*, returning error is  ls: cannot access ' /dev/video* ': No such file or directory.  I guess i need to write down a command like in this https://github.com/pjreddie/darknet/issues/222#issuecomment-333332899 but I don't know exactly how.

mive93, 2021-01-04:
I don't know, I worked with rtsp video stream, feeding them to tkDNN and it worked well just thanks to the OpenCV video capture.
How are you passing the input stream?

SokPhanith, 2021-04-24:
if you want detect video : ./demo yolo4_fp32.rt test.mp4 y
if you want detect webcam (jetson board) : ./demo yolo4_fp32.rt /dev/video0 y
- but something you need change port camera like /dev/video1 , /dev/video2.
- check camera on jetson board : ls /dev/video* if don't see anything mean that u don't have webcam connecting.
if you use picamera (jetson board) : ./demo yolo4_fp32.rt port y
     port = 'nvarguscamerasrc !  video/x-raw(memory:NVMM), width=3264, height=2464, format=NV12, 
     framerate=21/1 ! nvvidconv flip-method=0 ! video/x-raw, width=640, height=480, format=BGRx ! videoconvert ! 
   video/x-raw, format=BGR ! appsink'

closed 2021-11-25 by mive93