ceccocats/tkDNN #127: exec user process caused "exec format error

Hiroaki-K4, 2020-09-17:
Hi! Thank you for creating a useful library. 
When I tried to use tkDNN on docker and typed the following command, an error occurred. 

```
cd ~/tkDNN/docker
sudo docker build -t tkdnn:build -f Dockerfile .
```

How should I solve it? When I searched for the error, it said that I should add #! / Bin / bash to the entrypoint file, but is it correct?
Thanks!

```
nvidia@nvidia:~/tkDNN/docker$ sudo docker build -t tkdnn:build -f Dockerfile . 
[sudo] password for nvidia: 
Sending build context to Docker daemon  1.565MB
Step 1/3 : FROM ceccocats/tkdnn:latest
latest: Pulling from ceccocats/tkdnn
d7c3167c320d: Pull complete 
131f805ec7fd: Pull complete 
322ed380e680: Pull complete 
6ac240b13098: Pull complete 
006053f4160f: Pull complete 
7f180dbff330: Pull complete 
36adb6171cb0: Pull complete 
d7cf778701d8: Pull complete 
19ebebfc9562: Pull complete 
335ede31fbe9: Pull complete 
0d6c72ebfc16: Pull complete 
0211dac530d3: Pull complete 
e04f7e57fc62: Pull complete 
25f0770a8145: Pull complete 
8d5eabc9d0e3: Pull complete 
1fb7fbd0c6e2: Pull complete 
6a7c79fe938f: Pull complete 
fa3a50c7d9de: Pull complete 
afa98fb45a32: Pull complete 
20213feff8c8: Pull complete 
9a224688b140: Pull complete 
417656038cf1: Pull complete 
Digest: sha256:68e0359c215abecc9f0ca415271fdf264deaf974f5504c3a1b3a7593b36e5abe
Status: Downloaded newer image for ceccocats/tkdnn:latest
 ---> 1532824b3766
Step 2/3 : LABEL maintainer "Francesco Gatti"
 ---> Running in ee7893a6a198
Removing intermediate container ee7893a6a198
 ---> 37d4cc53c2b8
Step 3/3 : RUN cd && git clone https://github.com/ceccocats/tkDNN.git && cd tkDNN && mkdir build && cd build     && cmake .. && make -j12
 ---> Running in facbcbc8d84b
standard_init_linux.go:211: exec user process caused "exec format error"
The command '/bin/sh -c cd && git clone https://github.com/ceccocats/tkDNN.git && cd tkDNN && mkdir build && cd build     && cmake .. && make -j12' returned a non-zero code: 1
```

leizongxiong, 2020-10-11:
same to u, do you have solved？thanks

Hiroaki-K4, 2020-10-11:
I couldn't do it on docker, so I created an environment locally.

closed 2020-12-11 by Hiroaki-K4

ceccocats, 2020-12-11:
Hi sorry but I did not see this issue, you can't use this docker on NVIDIA Xavier, it is for x86_64 architecture

Hiroaki-K4, 2020-12-13:
ok. Thank you for reply!