ceccocats/tkDNN #128: mnist example

yaojiunn, 2020-09-17:
Your test program of mnist mnist_model.py only extract input.bin & output.bin. How to obtain c0.bin, c1.bin, d2.bin, d3.bin from my own training caffe model?

mive93, 2020-10-09:
Hi @yaojiunn 
Check here: https://github.com/ceccocats/tkDNN/blob/master/tests/exporters/caffe_weights_exporter.py

yaojiunn, 2020-12-01:
@mive93  Thank you for your answering. After running the python code you offered above, I got 8 bin files bellow:

Convolution0.bias.bin, Convolution0.bin, Convolution1.bias.bin, Convolution1.bin, 
InnerProduct2.bias.bin, InnerProduct2.bin, InnerProduct3.bias.bin, InnerProduct3.bin

I'm not sure which 4 bin files match the 4 bin files used in your example code (c0.bin, c1.bin, d2.bin, d3.bin).
Could you please give me a hint? Thank you!

Eugene-1995, 2020-12-17:
Hi @mive93, how to let the output bin files of caffe_weights_exporter.py (including weights & bias) fit your mnist sample code (seems no bias)?

mive93, 2021-01-04:
In tkDNN weights format bias and batchnorm are included in the convolutional weights, if that was the question.
Closing for now, feel free to reopen.

closed 2021-01-04 by mive93