ceccocats/tkDNN #129: Unable to Detect on ZED Camera / Converting sl::Mat to cv::Mat

Dadamsmeade, 2020-09-23:
Hi @ceccocats,
First off great work, I've really enjoyed using this library thus far, it's been extremely helpful.

I've run into a problem when trying use my ZED Camera on the Jetson Nano. 
Running the demo using /dev/video0 as raw video capture input works just fine, but I'd like to use ZED's API to grab depth info as well as the regular image so that I can display object distance. The main issue (I think) is converting frames from the ZED (sl::Mat) to cv::Mat. Fortunately, Stereolabs provides a useful walkthrough on how to use the ZED Camera with OpenCV here: https://www.stereolabs.com/docs/opencv/
When I apply this to the demo.cpp file provided in this repo, I can grab and retrieve frames from the ZED camera and display them via cv::imshow() no problem, but for some reason objects aren't detected when detNN->update is run, or at least they aren't drawn when displayed via cv::imshow(). 
This is how my code looks at the moment:

`#include <iostream>
#include <signal.h>
#include <stdlib.h>     /* srand, rand */
#include <unistd.h>
#include <mutex>

#include "CenternetDetection.h"
#include "MobilenetDetection.h"
#include "Yolo3Detection.h"

bool gRun;

void sig_handler(int signo) {
    std::cout<<"request gateway stop\n";
    gRun = false;
}

cv::Mat slMat2cvMat(Mat& input);

int main(int argc, char *argv[]) {

    std::cout<<"detection\n";
    signal(SIGINT, sig_handler);

    std::string net = "yolo4tiny_fp32.rt";
    //std::string input = "../demo/yolo_test.mp4";
    char ntype = 'y';
    int n_classes = 80;
    int n_batch = 1;
    bool show = true;
    float conf_thresh=0.3;    

    tk::dnn::Yolo3Detection yolo;
    tk::dnn::DetectionNN *detNN;  
   detNN = &yolo;

    detNN->init(net, n_classes, n_batch, conf_thresh);

    gRun = true;

    // Create a ZED camera object
    Camera zed;

    // Set configuration parameters
    InitParameters init_params;
    init_params.camera_resolution = RESOLUTION::HD1080;
    init_params.depth_mode = DEPTH_MODE::PERFORMANCE;
    init_params.coordinate_units = UNIT::METER;

   // Open the camera
    ERROR_CODE err = zed.open(init_params);
    if (err != ERROR_CODE::SUCCESS) {
        printf("%s\n", toString(err).c_str());
        zed.close();
        return 1; // Quit if an error occurred
    }

    // Set runtime parameters after opening the camera
    RuntimeParameters runtime_parameters;
    runtime_parameters.sensing_mode = SENSING_MODE::STANDARD;

    // Prepare new image size to retrieve half-resolution images
    Resolution image_size = zed.getCameraInformation().camera_resolution;
    int new_width = image_size.width / 2;
    int new_height = image_size.height / 2;

    Resolution new_image_size(new_width, new_height);

    // To share data between sl::Mat and cv::Mat, use slMat2cvMat()
    // Only the headers and pointer to the sl::Mat are copied, not the data itself
    Mat image_zed(new_width, new_height, MAT_TYPE::U8_C4);
    cv::Mat image_ocv = slMat2cvMat(image_zed);
    Mat depth_image_zed(new_width, new_height, MAT_TYPE::U8_C4);
    cv::Mat depth_image_ocv = slMat2cvMat(depth_image_zed);
    Mat point_cloud;

    cv::Mat frame;
    
    cv::namedWindow("detection", cv::WINDOW_NORMAL);

    std::vector<cv::Mat> batch_frame;
    std::vector<cv::Mat> batch_dnn_input;

    while(gRun) {
        batch_dnn_input.clear();
        batch_frame.clear();
        
        for(int bi=0; bi< n_batch; ++bi){
            if (zed.grab(runtime_parameters) == ERROR_CODE::SUCCESS) {

                // Retrieve the left image, depth image in half-resolution
                zed.retrieveImage(image_zed, VIEW::LEFT, MEM::CPU, new_image_size);
                zed.retrieveImage(depth_image_zed, VIEW::DEPTH, MEM::CPU, new_image_size);

                // Retrieve the RGBA point cloud in half-resolution
                // To learn how to manipulate and display point clouds, see Depth Sensing sample
                zed.retrieveMeasure(point_cloud, MEASURE::XYZRGBA, MEM::CPU, new_image_size);

                batch_frame.push_back(image_ocv);

                // this will be resized to the net format
                batch_dnn_input.push_back(image_ocv.clone());
            }
        } 
        if(!image_ocv.data) 
            break;
    
        //inference
        detNN->update(batch_dnn_input, n_batch);
        detNN->draw(batch_frame);

        if(show){
            for(int bi=0; bi< n_batch; ++bi){
                cv::imshow("detection", batch_frame[bi]);
                cv::waitKey(1);
            }
        }
    }

    std::cout<<"detection end\n";   
    double mean = 0; 
    
    std::cout<<COL_GREENB<<"\n\nTime stats:\n";
    std::cout<<"Min: "<<*std::min_element(detNN->stats.begin(), detNN->stats.end())/n_batch<<" ms\n";    
    std::cout<<"Max: "<<*std::max_element(detNN->stats.begin(), detNN->stats.end())/n_batch<<" ms\n";    
    for(int i=0; i<detNN->stats.size(); i++) mean += detNN->stats[i]; mean /= detNN->stats.size();
    std::cout<<"Avg: "<<mean/n_batch<<" ms\t"<<1000/(mean/n_batch)<<" FPS\n"<<COL_END;   

    return 0;
}

/**
* Conversion function between sl::Mat and cv::Mat
**/
cv::Mat slMat2cvMat(Mat& input) {
    // Mapping between MAT_TYPE and CV_TYPE
    int cv_type = -1;
    switch (input.getDataType()) {
        case MAT_TYPE::F32_C1: cv_type = CV_32FC1; break;
        case MAT_TYPE::F32_C2: cv_type = CV_32FC2; break;
        case MAT_TYPE::F32_C3: cv_type = CV_32FC3; break;
        case MAT_TYPE::F32_C4: cv_type = CV_32FC4; break;
        case MAT_TYPE::U8_C1: cv_type = CV_8UC1; break;
        case MAT_TYPE::U8_C2: cv_type = CV_8UC2; break;
        case MAT_TYPE::U8_C3: cv_type = CV_8UC3; break;
        case MAT_TYPE::U8_C4: cv_type = CV_8UC4; break;
        default: break;
    }

    // Since cv::Mat data requires a uchar* pointer, we get the uchar1 pointer from sl::Mat (getPtr<T>())
    // cv::Mat and sl::Mat will share a single memory structure
    return cv::Mat(input.getHeight(), input.getWidth(), cv_type, input.getPtr<sl::uchar1>(MEM::CPU));
}`

I've tried just about everything I can think of to figure this out, granted I'm fairly novice. I'm just wondering if anyone's had any experience with this or if anyone might know what the issue could be - maybe it's a memory issue? I'm not real familiar with how cpu/gpu memory is handled on the nano.

Thanks

anathema43, 2020-10-07:
@Dadamsmeade  may i know what fps are you getting for yolo3 (with int8) on nano?

Dadamsmeade, 2020-10-09:
@anathema43 I've only run yolo4_tiny, but the fps for that is pretty much all over the place when I run the code above, it's usually very high though (>300fps), perhaps because there's no inference is being done. If I use /dev/video0 as my input, however, everything works as expected and it hovers around 26fps

mive93, 2020-10-09:
Hi @Dadamsmeade 
sorry for the late answer. 

I haven't tried that, but I can guess where the problem is. 
First: did you check that the conversion was fine? That you got the same output. You said that you were able to show the image, so I guess so, but just to be sure. 
Second: Opencv mat are BGR and the code take that into account. What is the convention for  sl::Mat? Check the preprocess function (https://github.com/ceccocats/tkDNN/blob/master/src/Yolo3Detection.cpp) lines 86-86. 

Let me know if that was useful. If something else came up to my mind I will let you know.

Dadamsmeade, 2020-10-13:
Hi @mive93 ,
No worries about the delay!

> Opencv mat are BGR and the code take that into account. What is the convention for sl::Mat?
I did find that sl::Mat is in BGRA, so a cvtColor() to BGR is necessary, but when I initially applied this conversion it still wasn't running inference on the images, which is what led me to think it was maybe a memory issue.

Fortunately, I just stumbled upon a fix for this today, and I'm stoked, but I have to admit I don't really know _why_ it's fixed. I want to say it's a CUDA context thing, but I don't really know a lot about how the CPU/GPU utilize memory on Jetson platforms so I can't say for sure..

All I did was rearrange my main() function so that the ZED camera was opened before tk::dnn::DetectionNN *detNN was declared and initialized. Suddenly I could see bounding boxes on the output images!

I'll mark this as closed, but if you have any insight as to why this fixed it, I would love to hear it.

Thanks

closed 2020-10-13 by Dadamsmeade

anathema43, 2020-10-15:
@Dadamsmeade  please could you share the CmakeList.txt for zed with tkdnn and the cmake version your using.

[image]

here is my CMakeList.txt please let me know whats wrong
[CMakeLists.txt](https://github.com/ceccocats/tkDNN/files/5384494/CMakeLists.txt)