ceccocats/tkDNN #130: Export MobileNet-SSD Weights

Teresito, 2020-09-29:
Hi, you guys have an amazing repository. 

I am in need of guidance as I want to keep using your repository as it seems to be very substantial in executing fast inference times on the Jetson Nano board I have.

As suggested by the README, in order to get my custom trained MobileNet-SSD weights, I am to use the recommended [forked repository](https://github.com/mive93/pytorch-ssd) and export the produced debug and layer binary weights from there. However, I am in the dilemma where my custom trained MobileNet-SSD was done in TensorFlow and is in need of model conversion into PyTorch. 

Currently, I have spent more than a week doing my own attempts in converting and have failed (Attempts as in reading discussion forums and repositories that seems to conclude it isn't feasible). Should I consider doing the training  in PyTorch as a last resort?  Perhaps someone could shed me a light on this before I do so. 

Additionally, would producing my own .trt from my custom trained MobileNet-SSD and execute inference in your repository yield the same performance as any other .trt from your demo models?

Any discussion of ideas is much appreciated. Thank you :)

mive93, 2020-10-09:
Hi @Teresito 
I am actually more familiar with Pytorch rather then Tensorflow. I know there are some ways to port models from one to another framework,  however you have to be sure that the conversion is 100% correct. The implementation of Mobilenetv2-SSDLite that there is in this repo is specific to that model of the referenced pytorch repo. So yes, I would recommend training on that one to use the code as it is. Otherwise you can ofc change the code and make it work for your TF model, but that would require more time for you to get confidence with all tkDNN. 

For the performance, obviously it depends on many factors: the input size, the training dataset and so on. However, every layer would be as optimized as in the other models, so if would exploit the GPU at its maximum.

Teresito, 2020-10-19:
Hi @mive93 
Thank you for replying! I appreciate your feedback and I've taken your advice to try it out with the PyTorch training of your forked repository. I'll let you know in the future how it goes in the future.

Please feel free to open this issue for anyone who needs this same topic addressed again. I will be closing it for now.

closed 2020-10-19 by Teresito