ceccocats/tkDNN #132: RTSP Stream lagging

rafoliveira, 2020-10-01:
I am running a stream with tkDNN in Jetson Nano, but for some reason the detection skips two seconds, trying to keep up with the stream, every time they are out of sync.
I've tried to change the CAP_PROP_BUFFERSIZE variable using the following line:
`
cap.set(cv::CAP_PROP_BUFFERSIZE, 2);
`
However, this process doen't change its value.
Is this a issue in how the frame is read or a Jetson Nano performance bottleneck?
Thanks

mive93, 2020-10-09:
It is normal to skip frame if the elaboration can't keep up with the stream pace. 
What are the dimensions of the stream? 
Which network are you using? 
It could be that you are using a size too big or a network too slow for the Nano. 
However, a thing that you could to is decouple stream reading and elaboration using 2 separate threads. In that way, even if you skip some frames, you can be sure to always elaborate the last one.

rafoliveira, 2020-10-09:
Right, i will try the separate threads approach.
For now, i just reduce the frames in the stream from 25 to 15 and it solved the problem 90% of the time.

closed 2020-10-09 by rafoliveira