ceccocats/tkDNN #136: Failed to generate .rt for yolov4 with 'pwgen::PwgenException' error

Tsings04, 2020-10-13:
I have succeeded in generating the custom darknet-yolov3 to tensorrt format. But when i use the same environment and try to convert my custom darknet-yolov4 to .rt, it raises an error like this:

GPU free memory: 8399.88 mb.
New NetworkRT (TensorRT v7.13)
Float16 support: 1
Int8 support: 1
DLAs: 0
Selected maxBatchSize: 1
GPU free memory: 8311.8 mb.
Building tensorRT cuda engine...
**terminate called after throwing an instance of 'pwgen::PwgenException'
  what():  Driver error: 
Aborted (core dumped)**

Do you know how to solve that?

mive93, 2020-10-13:
Hi @Tsings04,
never had that error. 
Which board are you using? Have you tried rebooting? 

The only "related" thing I found is [this](https://forums.developer.nvidia.com/t/pwgenexception-when-building-a-cuda-engine/154469), but I don't think it will help you.

Tsings04, 2020-10-15:
> Hi @Tsings04,
> never had that error.
> Which board are you using? Have you tried rebooting?
> 
> The only "related" thing I found is [this](https://forums.developer.nvidia.com/t/pwgenexception-when-building-a-cuda-engine/154469), but I don't think it will help you.

I am working on my own PC with RTX 2080 Ti, and building the environment using Docker. The failed one i used is based on Ubuntu 16.04 with cuda 10.2 and cudnn 8. I switch to the other docker environment with ubuntu 18.04, cuda 10.2 and cudnn 7 and now it just works fine. I think i might be some conflicts between the docker env and my host env, my PC is on ubuntu 18.04 with cuda 10.2 and cudnn 7, it might be the reason of that. Anyways now it works for me, and thanks so much for your reply! :)

mive93, 2020-10-15:
Ok, nice to know :)
Closing it for now then, feel free to reopen.

closed 2020-10-15 by mive93