ceccocats/tkDNN #139: rtsp output

jaronski, 2020-10-19:
Hi all,
is it possible to suppress the output to display and instead stream the output of yolov4 ?

mive93, 2020-11-19:
Hi @jaronski ,

yeah, sure it's something duable.
To suppress the displayed output just put show to 0 (check show-flag https://github.com/ceccocats/tkDNN#run-the-demo).
To stream the output via rtsp, I have never done it but there should be some way maybe using Gstreamer or Opencv.
You should change the code of demo ([around here](https://github.com/ceccocats/tkDNN/blob/master/demo/demo/demo.cpp#L122)) once you have figured out how to stream with rtsp

biparnakroy, 2020-12-07:
@jaronski  so basically i need this streaming function too, so i tweaked the code and was able to do it, you can find the code here: https://github.com/biparnakroy/tkDNN, hope this helps you in your project.

closed 2021-01-04 by mive93