ceccocats/tkDNN #141: Cannot install CenterNet conda environment

Cram3r95, 2020-10-26:
Hi folks,

I have followed the instructions of tkDNN repository. Since I previosly worked with CenterNet and DeepSORT on my laptop, I have decided to install it in my Jetson AGX Xavier. Nevertheless, this repository says we have to follow the Install.md of the forked CenterNet repository.

Once I clone the repo and create a conda env with python 3.6, it fails in the step of installing torch 0.4.1:

 """
(CenterNet) robesafe@xavier:~/catkin_ws/src$ conda install pytorch=0.4.1 torchvision -c pytorch
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - torchvision
  - pytorch[version='>=0.4']
  - pytorch=0.4.1

Current channels:

  - https://conda.anaconda.org/pytorch/linux-aarch64
  - https://conda.anaconda.org/pytorch/noarch
  - https://conda.anaconda.org/c4aarch64/linux-aarch64
  - https://conda.anaconda.org/c4aarch64/noarch
  - https://conda.anaconda.org/archiarm/linux-aarch64
  - https://conda.anaconda.org/archiarm/noarch
  - https://conda.anaconda.org/conda-forge/linux-aarch64
  - https://conda.anaconda.org/conda-forge/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
""" 

And I cannot continue. I have observed that this pytorch version is old for Jetson purposes. What can i do?

ceccocats, 2020-11-09:
I suggest to train and test pytorch con your laptop/host pc and run only the inference with tkDNN on the Xavier.
Your issues is probably due to the different Architecture. Does it works on x86_64?

RobeSafe-UAH, 2020-11-09:
@ceccocats i have perfectly running that GitHub on my laptop and other computers, obviously x86_64, but i cannot install the required torch version in the Xavier embedded version. Do you mean to run the code and centernet dependencies but with another torch version in the Xavier? (The first version available is 1.1)

ceccocats, 2020-11-09:
If you want to run centernet with pytorch on Nvidia Xavier you should try to find the correct version for arch64 or upgrading the the 1.1 but you will probably need to fix something. 
Anyway this an issue more related to Centernet and not tkDNN.
tkDNN does not require pytorch, but only the weights that can be exported once with an x86 pc

RobeSafe-UAH, 2020-11-09:
@ceccocats mmm ... At the centernet section in the tkDNN repository, It is shown that you must install the centernet conda environment, but the requirements are for x86 arch, not for arm arch. Maybe It is not properly expressed.

mive93, 2020-11-10:
The Centernet section is there only if you want to train your own model on Centernet and then export your custom weights. Otherwise, if you want to use the one given by the authors, there is no need to use that.
When you run one of the centernet test (e.g. ./test_dla34_cnet, ./test_resnet101_cnet) it will automatically download the weights needed.
Therefore no anaconda env is required for tkDNN. Our centernet fork is used only to extract weights for tkDNN, for a custom train. 

I hope now it's more clear.

closed 2021-01-04 by mive93