ceccocats/tkDNN #143: TENSORRT LOG: CUDA initialization failure with error 2

panca1, 2020-11-09:
Hi everybody. 
I have an error when I run ./test_yolo4. Here is the code:

`Not supported field: batch=1
Not supported field: subdivisions=1
Not supported field: momentum=0.949
Not supported field: decay=0.0005
Not supported field: angle=0
Not supported field: saturation = 1.5
Not supported field: exposure = 1.5
Not supported field: hue=.1
Not supported field: learning_rate=0.00261
Not supported field: burn_in=1000
Not supported field: max_batches = 500500
Not supported field: policy=steps
Not supported field: steps=400000,450000
Not supported field: scales=.1,.1
Not supported field: mosaic=1
New NETWORK (tkDNN v0.5, CUDNN v8.004)
Reading weights: I=3 O=32 KERNEL=3x3x1
Reading weights: I=32 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=32 KERNEL=1x1x1
Reading weights: I=32 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=64 O=128 KERNEL=3x3x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=512 KERNEL=3x3x1
Reading weights: I=512 O=256 KERNEL=1x1x1
Reading weights: I=512 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=3x3x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=3x3x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=3x3x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=3x3x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=3x3x1
[... 353 lines cut ...]
260 ActivationLeaky   13 x   13,  512  ->   13 x   13,  512
261 Conv2d            13 x   13,  512  ->   13 x   13, 1024
262 ActivationLeaky   13 x   13, 1024  ->   13 x   13, 1024
263 Conv2d            13 x   13, 1024  ->   13 x   13,  512
264 ActivationLeaky   13 x   13,  512  ->   13 x   13,  512
265 Conv2d            13 x   13,  512  ->   13 x   13, 1024
266 ActivationLeaky   13 x   13, 1024  ->   13 x   13, 1024
267 Conv2d            13 x   13, 1024  ->   13 x   13,  255
268 Yolo              13 x   13,  255  ->   13 x   13,  255
===========================================================

GPU free memory: 25.428 mb.
New NetworkRT (TensorRT v7.21)
TENSORRT LOG: CUDA initialization failure with error 2. Please check your CUDA installation:  http://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html
Errore di segmentazione (core dump creato)
`
On my laptop I have Ubuntu 18.04, kernel version 5.4.0, gcc version 8.4, GLIBC 2.27, Cuda11.1, cuDNN8.0.4, Tensort7.2.1, openCV 4.4.0

So I think I have all the requirements that cuda11.1 and tkDNN needs.

I have tried also to reinstall all of these, but I obtain anyway the error.

cmake .. and make works.

How can I fix that?

panca1, 2020-11-09:
I've also tried now to install cuda 10.0, cuDNN 7.6, Tensorrt 6.0.1 but it give me back the same problem? 
Is it a problem of out-of-memory?

mive93, 2020-11-09:
Hi @panca1 , 

the problem could be with CUDA version of the libraries.
Run these commands and check that cuda is always the same, as in my case (check image, always CUDA 10.2).

```
dpkg -l | grep cudnn
dpkg -l | grep nvinfer
nvcc --version
```

[image]

panca1, 2020-11-09:
Yes, this is the problem. 

[image]

panca1, 2020-11-10:
Hi @mive93 ,
I fix the problem but it give me back that.

`TENSORRT LOG: Generating copy for Upsample214_out to Route218_out
TENSORRT LOG: Eliminating concatenation Route233
TENSORRT LOG: Retargeting ActivationLeaky228_out to Route233_out
TENSORRT LOG: Eliminating concatenation Route236
TENSORRT LOG: Retargeting ActivationLeaky235_out to Route236_out
TENSORRT LOG: Retargeting ActivationLeaky211_out to Route236_out
TENSORRT LOG: Eliminating concatenation Route251
TENSORRT LOG: Retargeting ActivationLeaky246_out to Route251_out
TENSORRT LOG: Eliminating concatenation Route254
TENSORRT LOG: Retargeting ActivationLeaky253_out to Route254_out
TENSORRT LOG: Retargeting ActivationLeaky194_out to Route254_out
TENSORRT LOG: After concat removal: 273 layers
TENSORRT LOG: Graph construction and optimization completed in 0.238891 seconds.
TENSORRT LOG: TensorRT was linked against cuDNN 7.6.3 but loaded cuDNN 7.6.0
TENSORRT LOG: Constructing optimization profile number 0 out of 1
*************** Autotuning format combination: Float(1,416,173056,519168) -> Float(1,416,173056,5537792) ***************
TENSORRT LOG: ../rtSafe/safeRuntime.cpp (25) - Cuda Error in allocate: 2 (out of memory)
TENSORRT LOG: ../rtSafe/safeRuntime.cpp (25) - Cuda Error in allocate: 2 (out of memory)
cloud not build cuda engine
/home/giacomo/Scrivania/tkDNN/src/NetworkRT.cpp:145
Aborting...
`

Is it a problem of GPU memory?

mive93, 2020-11-16:
Yes, as the debug says, it is an out of memory problem.

closed 2020-11-26 by panca1