ceccocats/tkDNN #144: CUDNN failure CUDNN_STATUS_BAD_PARAM: Conv2d.cpp: 105

xrobellet, 2020-11-10:
Hi,
I am testing your repo on a jetson nano (flash with jetpack 4.4).
Compilation works fine but when I try to construct the engine.rt (whatever yolo version I am testing) I receive a message error:
 CUDNN failure CUDNN_STATUS_BAD_PARAM: Conv2d.cpp: 105
The engine.rt is there at the end and seems to work but I am concern that error could decrease the inference.
Could you please give me any feeback on this problem?
thanks a lot

mive93, 2020-11-16:
Hi @xrobellet 
I haven't fully get your problem.
So you are able to create the rt file but at the end you get CUDNN_STATUS_BAD_PARAM?
To create the trt on the Nano have you followed our solution explained in https://github.com/ceccocats/tkDNN/issues/26 ?
In case, it's normal and there is no problem, you should be able to then use the rt file correctly. 
Have you tested with the demo?
Otherwise, I don't know, I would need more infos.

xrobellet, 2020-11-16:
Hi,
thank a lot for your answer. Yes I followed all the recommandations from  the solution explained in #26. 
I attached a screenshot when I construct the engine.rt. despite the fact it's say aborting.. a .rt engine is created and seems to work fine on the demo. I just concern about this weird message "CUDNN failure CUDNN_STATUS_BAD_PARAM: Conv2d.cpp: 105 "

[image]

mive93, 2020-11-16:
Cool :) 
Then it's fine. The problem is just because the code we suggested inserting frees some memory and doing so corrupts the CUDDN model, but the resulting rt model is correct. So everything is fine :+1:

xrobellet, 2020-11-16:
cool thx a lot !!!

closed 2020-11-16 by xrobellet