ceccocats/tkDNN #147: Custom model, this is not yolo3 error

alexflorensa, 2020-11-12:
Hello @ceccocats,

I am trying to make inference with a custom model using tensorrt + tkdnn.
1. To build trt engine, I've been using `demo_darknet2onnx.py` script of this [repo](https://github.com/Tianxiaomo/pytorch-YOLOv4 ), so once I got my ONNX model I executed: 
```trtexec onnx=yolov4_1_3_416_416_static.onnx --explicitBatch --saveEngine=yolo_custom.trt```

2. Now, having trt engine, I cloned the darknet repo from gitlab (mentioned in the README) to get debug and layers folders as the yolov4 example from your readme (doing the export)

3. With trt engine built and the darknet layers exported, I tried to make predictions executing: ./demo yolo_custom.trt
At this step I got an unexpected error: **"this is not yolo 3"**

Do you know what can I do? Should I build my trt engine in another way? How did you build it?
Thank you.

closed 2020-11-14 by alexflorensa

Goru1890, 2020-11-26:
How did you resolve it?

alexflorensa, 2020-11-26:
Running it on the docker. They have an image on dockerhub or you can build
it with their Dockerfile. Maybe was a version problem: not matching their
requirements.

AGpix, 2021-05-05:
Hi, how did you resolve this issue