ceccocats/tkDNN #148: tkDNN only for testing?

Goru1890, 2020-11-18:
I'm not understanding it at all. Does tkDNN only acellerates the testing of a trained yolo weight?
So if I want to test tkDNN with yolo with my custom data I have to train first in darknet with a yolo cfg with batch 1-4 (subdivision=1). And than parse the cfg and the optained weight file to bin files?

mive93, 2020-11-18:
Hi @Goru1890,
Yes, tkDNN is application oriented, so it optimizes only the inference to improve the performance of a real application. 
So yes, you should first train your model with another framework (as darknet) and then export the weights in tkDNN format and use it for boosting the performance on NVIDIA boards.
As written in the readme, we have a fork of darknet where we implemented the export of the weights. So once you have your cfg and .weights file, you can export the weights for tkDNN via that fork and then create a test on tkDNN.

Goru1890, 2020-11-18:
Ok, thank you very much. 
Do I have to change some parameters in [net] section in the darknet cfg file for training such as batch size?
I read the readme from this repo and Alexey repo and it's seems you always use a batch size between 1 and 4, do I have to train also with that sizes?

mive93, 2020-11-18:
It does not matter. Those batch sizes are two different stuff.
The batch size that you use in the training is how many images you are loading for the training phase. It is a hyper-parameter of the training process and it affects it. 
The batch size that you use for inference is still how many images you are loading, but to perform inference onto. It changes the throughput of your system in the end (at the same time you inference on 4 images instead of just one), but of course it depends if your application needs that or not.

Hope it's clear now.

Goru1890, 2020-11-19:
Is there a difference declaring batch size in cfg file or with the command export TKDNN_BATCHSIZE? 

Should I always declare batch size = 1 in cfg and then if I want to use a different size may I use the command?

mive93, 2020-11-19:
The batch size in the cfg does not affect tkDNN at all. 
If you want to create a rt file with a batch size greater than one (e.g.4), then you have to do 
```
export TKDNN_BATCHSIZE=4
```
And your rt model will support batch sizes from 1 to 4 (but it will be optimized for 4).
Then if you want to use it in the demo you have to pass the correct number of batches that you want to use.

Goru1890, 2020-11-19:
So can I export into bin files the same cfg file that I used for trainging witch batch= 64 and subdivision=16?
And then use 

`export TKDNN_BATCHSIZE=4`

I tried that yesterday and the export didn't work, I read in some issue that using batch= 1 would resolve the problem and it worked.

mive93, 2020-11-19:
Yes, when you export the weights just set batch and subdivision to 1 (as in testing).
Then you have the weights ready for tkDNN. They will work with any inference batch size (again, training and inference batch sizes are not related). 

So it depends on you,  the memory of your GPU, and the needs of your application. 
If you don't need batching, don't set it. If you need it, then set the environment variable TKDNN_BATCHSIZE before lunching the test (and delete the old rt net if you already have one).

Goru1890, 2020-11-19:
Thank you very much for your time and help.

mive93, 2020-11-19:
You are welcome.
I will close the issue for now.

closed 2020-11-19 by mive93