ceccocats/tkDNN #149: Build Error: ‘blobFromImage’ is not a member of ‘cv::dnn’

Yuanye-F, 2020-11-24:
[ 26%] Building CXX object CMakeFiles/tkDNN.dir/src/Int8BatchStream.cpp.o
/home/user/Downloads/tkDNN/src/Int8BatchStream.cpp: In member function ‘void BatchStream::readCVimage(std::string, std::vector<float>&, bool)’:
/home/user/Downloads/tkDNN/src/Int8BatchStream.cpp:129:33: error: ‘blobFromImage’ is not a member of ‘cv::dnn’
     m_LetterboxImage = cv::dnn::blobFromImage(m_LetterboxImage);
                                 ^~~~~~~~~~~~~
make[2]: *** [CMakeFiles/tkDNN.dir/src/Int8BatchStream.cpp.o] Error 1
make[1]: *** [CMakeFiles/tkDNN.dir/all] Error 2
make: *** [all] Error 2

while there is no problem for cmake:
-- The C compiler identification is GNU 7.3.1
-- The CXX compiler identification is GNU 7.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/rh/devtoolset-7/root/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/rh/devtoolset-7/root/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda (found suitable version "10.2", minimum required is "9.0") 
-- Found CUDNN: /usr/local/cuda/lib64/libcudnn.so
-- Found CUDNN include: /usr/local/cuda/include
-- Found NVINFER: /usr/local/cuda/lib64/libnvinfer.so
-- Found NVINFER include: /usr/local/cuda/include
-- Found CUDNN: /usr/local/cuda/lib64/libcudnn.so  
install dir:/usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/Downloads/tkDNN/build

Please help, thanks a lot : )

mive93, 2020-11-24:
Hi @Yuanye-F 

Do you have opencv installed?
In case, which version?
[Here](https://github.com/ceccocats/tkDNN/blob/master/scripts/install_OpenCV4.sh) you can find the script to install the required one.

Yuanye-F, 2020-11-24:
Hi @mive93 

I have installed opencv and the version is 4.1.2.

mive93, 2020-12-04:
I don't know, the problem is clearly related to OpenCV. 
You have to make sure that it's linked correctly. 
Is opencv installed in /usr/local/ ?

Yuanye-F, 2020-12-07:
Thanks.  I would check if there is something wrong with OpenCV.
OpenCV is installed in /usr/local/

mive93, 2021-01-04:
Then I don't know what to suggest. 
Are you working on a computer or embedded board?

irvingzhang0512, 2021-05-08:
Met the same issue, try add `-lopencv_dnn` in `tkdnn_LIBS`

`set(tkdnn_LIBS kernels ${CUDA_LIBRARIES} ${CUDA_CUBLAS_LIBRARIES} ${CUDNN_LIBRARIES} ${OpenCV_LIBS} yaml-cpp -lopencv_dnn -lopencv_videoio)`

Yuanye-F, 2021-05-10:
> Met the same issue, try add `-lopencv_dnn` in `tkdnn_LIBS`
> 
> `set(tkdnn_LIBS kernels ${CUDA_LIBRARIES} ${CUDA_CUBLAS_LIBRARIES} ${CUDNN_LIBRARIES} ${OpenCV_LIBS} yaml-cpp -lopencv_dnn -lopencv_videoio)`

Get it. Thanks a lot.

Yuanye-F, 2021-05-10:
@irvingzhang0512 
Get it. Thanks a lot.

closed 2021-05-10 by Yuanye-F