ceccocats/tkDNN #150: how export debug and

xn1997, 2020-11-25:


mive93, 2020-11-25:
Hi @xn1997 
what do you mean? 
The weights are for tkDNN (which is in C++), not for pytorch.

xn1997, 2020-11-25:
> Hi @xn1997
> what do you mean?
> The weights are for tkDNN (which is in C++), not for pytorch.

sorry, I know how to translate the weight to .rt for tkDNN, but I don't know how to export the weights from pytorch.
is there any example on how to get the weight (debug and layers folders) from pytorch model?

mive93, 2020-11-25:
Oh I get it now.
Yeah, you can find 3 examples:
- for centernet here: https://github.com/sapienzadavide/CenterNet/blob/master/src/lib/models/exporter.py
- for mobilenet here: https://github.com/mive93/pytorch-ssd/blob/master/run_ssd_live_demo.py
- for shelfnet here: https://git.hipert.unimore.it/mverucchi/shelfnet/blob/citys_lw/ShelfNet18_realtime/export.py

As you can see it's almost the same code, but it's a bit adapted to the type of layers of the net.
Unfortunately we have not written yet a general exporter from PyTorch, but those examples should help.

xn1997, 2020-11-26:
> Oh I get it now.
> Yeah, you can find 3 examples:
>
>     * for centernet here: https://github.com/sapienzadavide/CenterNet/blob/master/src/lib/models/exporter.py
>
>     * for mobilenet here: [...]

Thank you! : )

closed 2021-01-04 by mive93