ceccocats/tkDNN #154: input_dim doesn't match buffersDIM[0]

nightduck, 2020-11-27:
I built a tensorrt engine with a custom architecture and I'm using the NetworkRT class to simplify inference.

In the NetworkRT class, the dimensions of my input_dim are {n=1, c=0, h=0, w=0, l=1}, which are obviously incorrect. An input image can't be 0x0 pixels. But in buffersDIM[0], the dimensions are correct: {n=1, c=1, h=192, w=192, l=1}.

Similar story for output_dim and buffersDIM[0]: {n=1, c=0, h=0, w=0, l=1} and {n=1, c=1, h=2, w=0, l=1}, respectively.

Is this a problem with my TensorRT engine, or am I missing something when initializing my NetworkRT instance?

nightduck, 2020-12-05:
The problem lies in [these two lines](https://github.com/ceccocats/tkDNN/blob/b8855b9599e52a51b371e99255063cd6f00fecd7/src/NetworkRT.cpp#L164) in NetworkRT.cpp

    buf_input_idx = engineRT->getBindingIndex("data"); 
    buf_output_idx = engineRT->getBindingIndex("out");

But since I'm using a custom architecture, my input names don't match "data" and "out", so the input and output indices are never found. This causes a cascade of silent errors which evenutally manifest when input_dim is the wrong dimensions, or (if that's manually set), the output buffers are never initialized.

ceccocats, 2020-12-10:
You need to know the name of the input and output layer of the network and set it to the getBindingIndex function.
You can iterate the layers using the engineRT object but unfortunately I don't think there is an example in tkDNN

nightduck, 2020-12-11:
Even if I known the name of my layers (I do), I have to modify NetworkRT.cpp in order to make use of them. So the next time I git pull the tkDNN repo, this bug comes back.