ceccocats/tkDNN #158: FPS Results using TX2 - Jetpack 4.4

andressasuiti, 2020-12-02:
Hi! I would like to know if you have any FPS results from yolov4 using Nvidia TX2, but with Jetpack 4.4. I trained my own network and the results I got were between 1 and 3 FPS slower than what you got with Jetpack 4.2

mive93, 2020-12-04:
Hi @andressasuiti 
Can you give more details?
With which model, with which mode (FP32, FP16)? Have you enabled jetson_clock?

garyzf, 2020-12-08:
I get 86-87 fps using nvidia tx2

garyzf, 2020-12-08:
yolov4-tiny(320*320) , with Jetpack 4.2

garyzf, 2020-12-08:
fp32,B=1

andressasuiti, 2020-12-08:
Hi @mive93 
I enabled jetson_clock -m 0 and tested the following resolutions: 416x416, 512x512, 608x608, and 736x736, both with FP32 and FP16. My fastest result was around 9 FPS and the slowest was 2.5 FPS. This using yolov4 and batch size 1. Is it normal?

MuhammadAsadJaved, 2020-12-14:
Hi, Can anyone explain how we can know about the FPS during demo?  It only show the output video but does not print FPS, 

I am using Yolov3  using `./demo`

Thanks

mive93, 2021-01-04:
@andressasuiti then I don't know. [Here](https://git.hipert.unimore.it/edbench/edbench/tree/master) you can find further and deeper results for Tx2 with Jetpak 4.4, where the FPS are real ones (end2end, not only inference). Hope it helps.

@MuhammadAsadJaved, just let the demo finish (or stop it with CTRL+C) and you will get the numbers.

closed 2021-11-25 by mive93