ceccocats/tkDNN #159: Cmake error

lxyzler, 2020-12-04:
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_nvinfer_LIBRARY

How to deal with the problem.

mive93, 2020-12-04:
Hi @lxyzler 

check to have correctly installed all the required libraries.
At the end, you should have something like this: https://github.com/ceccocats/tkDNN/issues/143#issuecomment-724269903.
Then check all the other issues on cmake (there are plenty)

lxyzler, 2020-12-07:
This problem has been solved. There are new problems
[image]

mive93, 2021-01-04:
Now you have a problem with the other required libraries for this project, namely opencv and yamlcpp. 
Have you doubled check the readme, dependency section?

qrzeffort, 2021-02-05:
When cmake.. I had a similar problem:
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_cublas_device_LIBRARY (ADVANCED)
    linked by target "tkDNN" in directory /home/nvidia/tkDNN

The dependency:
cudnn=7.3.1
opencv=3.3.1
cuda=10.0
I followed the steps in readme,

liberize, 2021-02-05:
@qrzeffort
follow https://gitlab.kitware.com/cmake/cmake/-/merge_requests/2298/diffs?commit_id=bdf1f361354e3e0ecd4fd5ede2f38d5384a0444a
patch /usr/share/cmake-3.10/Modules/FindCUDA.cmake yourself

qrzeffort, 2021-02-05:
> @qrzeffort
> follow https://gitlab.kitware.com/cmake/cmake/-/merge_requests/2298/diffs?commit_id=bdf1f361354e3e0ecd4fd5ede2f38d5384a0444a
> patch /usr/share/cmake-3.10/Modules/FindCUDA.cmake yourself

Thank you very much！

PhuongNDVN, 2021-06-15:
@lxyzler I get the same problem. Could you share how to fix the problem?

mive93, 2021-08-19:
Which version of cmake are you using? Minimum 3.15 is required.

closed 2021-11-25 by mive93