ceccocats/tkDNN #160: Do you have any plan to support EfficientDet series?

TPFRL, 2020-12-04:
Thanks for making this awesome project!

It will be really great if EfficientDet-series are available on tkDNN platform.
As I understand, three additional features need to be implemented:
1. swish activation function
2. [scale_channels] layer 
3. [shortcut] layer with [weighted multi input](https://github.com/AlexeyAB/darknet/issues/4662)

Do you have any plan to support any of these features?

mive93, 2020-12-04:
Hi @TPFRL 
Thank you!
For now, we do not have in plan to add them, however it could be possible that in the future we'll support them.
Generally, we port the thing that we need for the projects we work on. 

In any case, thank you for your notes!
I will let you know if we'll start working on that.

chjej202, 2020-12-15:
Hi @mive93 

Somebody created a new pull request  (#172) which implements scale_channels and swish activation to support enet-coco.cfg 

How about merge this pull request to your branch?

mive93, 2021-01-04:
Hi @chjej202 
we have many things to merge to the master (segmentation networks, 3d object detection, tracking), but we have several things to do first at work. As soon as we'll have time, we'll consider that also. 
Thanks for the remainder in any case.