ceccocats/tkDNN #162: Yolo3Detection never allocates input_d (was working a week ago)

nightduck, 2020-12-04:
My program creates a Yolo3Detection network, and crashes after calling `detNN->update(frames)`.

The error finally lands in YoloRT.h's function entry_index, where there's a division by zero error.

But after debuggin the the update function, I found at this line:

    netRT->infer(dim, input_d);

`input_d` is a null pointer. What could cause this? Anywhere else I can look for clues? This was working a week ago.

nightduck, 2020-12-04:
Confirmed this bug is not present in 86478f9384eef13d68a9406ee12fbcb4df6ab892

ceccocats, 2020-12-10:
Hi, we updated the structure of YoloRT.h to have more parameters, this parameters are serialised on the .rt file.
So if you haven't rebuilt the RT it is outdated.
Rebuild the RT should be sufficient

nightduck, 2020-12-21:
Rebuilt the RT file and now it's complaining "this is not yolo3" and throwing a fatal error. In its defense, it's right: I'm feeding it yolo4. But I was previously able to load a yolo4 tensorRT engine into a Yolo3Detection object. Is this no longer supported?

nightduck, 2020-12-22:
Rebuilding the RT file doesn't help. Results of my debugging.

The YoloRT is used during inference. But the "configure" method is never called, so by the time "enqueue" is called during inference, the "h" and "w" fields are still zero, resulting in a division by zero error.

I can't manually call it because the YoloRT instance is hidden inside an abstracted runtime context. Is this still something wrong with the RT file?

nightduck, 2020-12-22:
I just followed @ceccocats's advice over and over again, starting over each time until eventually it worked. As for why it didn't work to begin with, I'm going to assume it's a classic case of PEBCAK.

closed 2020-12-22 by nightduck