ceccocats/tkDNN #164: Error reading file yolo4tiny/layers/c2.bin with n of float: 36864 seek: 0 size: 147456

winterxx, 2020-12-07:
I change the yolo4tiny input size to 608x608 and train my yolo4tiny model, export weights is correct, but creat the .rt file is error by running: ./test_yolo4tiny.
the error:
New NETWORK (tkDNN v0.5, CUDNN v7.604)
Reading weights: I=3 O=32 KERNEL=3x3x1
Reading weights: I=32 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=3x3x1
Error reading file yolo4tiny/layers/c2.bin with n of float: 36864 seek: 0 size: 147456

/media/xx/data1/projects/project2020/yolo/yolotools/tkDNN/src/utils.cpp:58
Aborting...

winterxx, 2020-12-07:
> I change the yolov4-tiny-custom.cfg input size to 608x608 and train my yolo4tiny-custom model, export weights is correct, but creat the .rt file is error by running: ./test_yolo4tiny.
> the error:
> New NETWORK (tkDNN v0.5, CUDNN v7.604)
> Reading weights: I=3 O=32 KERNEL=3x3x1
> Reading weights: I=32 O=64 KERNEL=3x3x1
> Reading weights: I=64 O=64 KERNEL=3x3x1
> Error reading file yolo4tiny/layers/c2.bin with n of float: 36864 seek: 0 size: 147456
> 
> /media/xx/data1/projects/project2020/yolo/yolotools/tkDNN/src/utils.cpp:58
> Aborting...

closed 2020-12-07 by winterxx

poornimajd, 2021-02-23:
> I change the yolov4-tiny-custom.cfg input size to 608x608 and train my yolo4tiny-custom model, export weights is correct, but creat the .rt file is error by running: ./test_yolo4tiny.
> the error:
> New [...]

@winterxx How did you solve it?