ceccocats/tkDNN #165: darknet yolov4x-mish.weights export layers error

toplinuxsir, 2020-12-09:
darknet yolov4x-mish.weights export layers error
```
./darknet export ./yolov4x-mish.cfg ./yolov4x-mish_best.weights  layers
```

error log
```
Loading weights from ./yolov4x-mish_best.weights...
 seen 64, trained: 785 K-images (12 Kilo-batches_64) 
Done! Loaded 203 layers from weights-file 
n: 0, type 0
Convolutional
weights: 864, biases: 32, batch_normalize: 1, groups: 1
write binary layers/c0.bin
fish: “./darknet export ./yolov4x-mish…” terminated by signal SIGSEGV (Address boundary error)
```

thnkinbtfly, 2020-12-14:
Did you made the 'layers' directory?
https://github.com/ceccocats/tkDNN#1export-weights-from-darknet

toplinuxsir, 2020-12-14:
@thnkinbtfly  Thanks  , I fixed it .

mive93, 2021-01-04:
Thanks @thnkinbtfly, that was correct.
Closing for now, feel free to reopen.

closed 2021-01-04 by mive93