ceccocats/tkDNN #167: different BN epsilon from darknet

thnkinbtfly, 2020-12-11:
Hi, thanks for this great project!!

I was debugging some layers outputs and tried to find the reason why some are different...

I think the default epsilon used in tkDNN (1e-5) is different from darknet (1e-6).

Darknet uses 1e-6, as below.
https://github.com/AlexeyAB/darknet/blob/6be487b51f720fad926b6e2fff5685b3328561c5/src/blas.c#L292

But tkDNN is using 1e-5.
https://github.com/ceccocats/tkDNN/blob/b8855b9599e52a51b371e99255063cd6f00fecd7/include/tkDNN/Layer.h#L35

After updating this, the difference I confronted was resolved!
Could you check this?

thnkinbtfly, 2020-12-11:
I just checked the original darknet changed the epsilon to 1e-5.
https://github.com/AlexeyAB/darknet/commit/a34682e67c9ff279019e17070a5d34d596cd8422

Could you consider using same epsilon when exporting the weight and running the network?

mive93, 2021-01-04:
Hi @thnkinbtfly 
I don't get the point.
So you're aying that he's using 1e-6 or 1e-5?
And where do you experienced the error?

closed 2021-11-25 by mive93