ceccocats/tkDNN #169: Segmentation fault (core dumped)

tim94173, 2020-12-12:
I run `./darknet export <cfg> <weights>`but it  show

nms_kind: greedynms (1), beta = 0.600000 
Total BFLOPS 10.903 
avg_outputs = 276828 
 Allocate additional workspace_size = 19.66 MB 
Loading weights from backup/yolov4-tiny3l-test1_16000.weights...
 seen 64, trained: 1024 K-images (16 Kilo-batches_64) 
Done! Loaded 50 layers from weights-file 
n: 0, type 0
Convolutional
weights: 864, biases: 32, batch_normalize: 1, groups: 1
write binary (null)/c0.bin
Segmentation fault (core dumped)

how can i fix it?

thnkinbtfly, 2020-12-13:
I think you missed the instructions.
If you do mkdir and add another argument, I think it will be resolved.

```
mkdir layers debug
./darknet export <path-to-cfg-file> <path-to-weights> layers
```
https://github.com/ceccocats/tkDNN#1export-weights-from-darknet

tim94173, 2020-12-14:
I am an idiot , I forgot enter "layers" haha...

closed 2021-01-04 by mive93