ceccocats/tkDNN #177: Run yolov4tiny demo error

lovehuanhuan, 2020-12-16:
win10 rtx2080ti  nvidiadocker

root@04572d63d538:/yolo/tkDNN/build# ./demo yolo4tiny_fp32.rt ../demo/yolo_test.mp4 y 1 1 0
detection
yolo4tiny_fp32.rt
New NetworkRT (TensorRT v7)
Float16 support: 1
Int8 support: 1
DLAs: 0
create execution context
Input/outputs numbers: 3
input index = 0 -> output index = 2
Data dim: 1 3 416 416 1
Data dim: 1 255 26 26 1
RtBuffer 0   dim: Data dim: 1 3 416 416 1
RtBuffer 1   dim: Data dim: 1 255 13 13 1
RtBuffer 2   dim: Data dim: 1 255 26 26 1
Unable to stop the stream: Inappropriate ioctl for device
detection end

Time stats:
Segmentation fault

mive93, 2021-01-04:
Again, we do not support Windows right now, I'm sorry

mive93, 2021-11-25:
We do now support windows.
Closing for inactivity. Feel free to reopen.

closed 2021-11-25 by mive93