ceccocats/tkDNN #178: I want to convert a darknet model trained on 4 channels to an rt file

Hiroaki-K4, 2020-12-16:
Hi.
I want to convert a darknet model trained on 4 channels to an rt file.
I modified the inside of darknet to train the model in 4 channels with rgb plus depth. So I tried exporting the model in the link repository, but a segmentation fault occurred.
https://git.hipert.unimore.it/fgatti/darknet
Where should I change?

Hiroaki-K4, 2020-12-16:
I solved this problem by commenting out network_predict (net, X) in darknet.c.
However, an error occurred while doing ./test_yolo4 of tkDNN. The error message is as follows.

```
Not supported field: batch=64
Not supported field: subdivisions=64
Not supported field: momentum=0.949
Not supported field: decay=0.0005
Not supported field: angle=0
Not supported field: saturation = 1.5
Not supported field: exposure = 1.5
Not supported field: hue=.1
Not supported field: learning_rate=0.001
Not supported field: burn_in=1000
Not supported field: max_batches = 500500
Not supported field: policy=steps
Not supported field: steps=400000,450000
Not supported field: scales=.1,.1
Not supported field: mosaic=1
New NETWORK (tkDNN v0.5, CUDNN v7.603)
Reading weights: I=4 O=32 KERNEL=3x3x1
Reading weights: I=32 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=32 KERNEL=1x1x1
Reading weights: I=32 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=64 O=128 KERNEL=3x3x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=512 KERNEL=3x3x1
Reading weights: I=512 O=256 KERNEL=1x1x1
Reading weights: I=512 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=3x3x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=3x3x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=3x3x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=3x3x1
Reading weights: I=256 O=256 KERNEL=1x1x1
[... 362 lines cut ...]

GPU free memory: 4444.99 mb.
New NetworkRT (TensorRT v6.01)
Float16 support: 1
Int8 support: 1
DLAs: 2
Selected maxBatchSize: 1
GPU free memory: 4136.89 mb.
Building tensorRT cuda engine...
serialize net
create execution context
Input/outputs numbers: 4
input index = 0 -> output index = 3
Data dim: 1 4 512 512 1
Data dim: 1 54 16 16 1
RtBuffer 0   dim: Data dim: 1 4 512 512 1
RtBuffer 1   dim: Data dim: 1 54 64 64 1
RtBuffer 2   dim: Data dim: 1 54 32 32 1
RtBuffer 3   dim: Data dim: 1 54 16 16 1
Error reading file yolo4/layers/input.bin with n of float: 1048576 seek: 0 size: 4194304

/home/nvidia/tkDNN/src/utils.cpp:58
Aborting...
```
Thanks!

Hiroaki-K4, 2020-12-16:
I solved it by comment out test_inference.

Hiroaki-K4, 2020-12-17:
Hi. I could convert weights file to rt file.
But my 4 channel model can't detect anything. Does the conversion method support 4 channels?
Thanks!

mive93, 2021-01-04:
Hi @Hiroaki-K4 
Do you still have the problem?
It should support that.

Hiroaki-K4, 2021-01-05:
Hi @mive93 
Yes, I have the problem.
Do I need to change the source code?

poornimajd, 2021-02-23:
Hi @Hiroaki-K4
How did you solve the issue?

Hiroaki-K4, 2021-02-23:
@poornimajd 
No, it hasn't been resolved yet.

poornimajd, 2021-02-25:
@Hiroaki-K4 
I think it is a dependency issue.Was able to solve it when tried it on rtx 2080 ti with cuda 10.

Hiroaki-K4, 2021-02-25:
@poornimajd 
oh, really?
Did you change either tkDNN or somewhere in the [darknet](https://git.hipert.unimore.it/fgatti/darknet) that exports weights?

poornimajd, 2021-02-25:
The problem was with creation of the layers and debug folder.When I created these folders in rtx 2080 and then used these in gtx 1060 (where the error occurred earlier) with [tkdnn](https://github.com/ceccocats/tkDNN) ,and also tried with [tkdnn](https://github.com/ioir123ju/tkDNN) I was able to get the trt model for fp32,fp16 and int8 in both the repos.

Hiroaki-K4, 2021-02-25:
@poornimajd 
ok. Is it possible to detect the expected object with the created rt file?

poornimajd, 2021-02-25:
Yes it is possible.

Hiroaki-K4, 2021-02-26:
@poornimajd  Oh, Great!
Did you comment out what I said earlier in this issue?

poornimajd, 2021-02-26:
No.

Hiroaki-K4, 2021-02-26:
@poornimajd 
ok, thank you.
I will try it again.

closed 2021-03-24 by Hiroaki-K4

najingligong1111, 2021-05-24:
Have you solved the problem when input channel is 4 !

Hiroaki-K4, 2021-05-24:
@najingligong1111 
yes, I have solved.
You need to comment out network_predict (net, X) in darknet.c when you export darknet weights file.
You also have to comment out below line in yolo4.cpp when you run ./test_yolo4.

```
int ret = testInference(input_bins, output_bins, net, netRT);
```