ceccocats/tkDNN #179: difference of yolov4x-mish.cfg between yours and Alexey

erikguo, 2020-12-22:
hi, thank you for your great job!

I found there's one difference of yolov4x-mish.cfg between yours and Alexey. 

The activation func just before yolo layer is linear in your yolov4x-mish.cfg while the func is logistic. Why do you change the activation func? 

@ceccocats

mive93, 2021-01-04:
Hi @erikguo 
We used the cfg file that Alexey gave us when he firstly released yolov4x-mish. So, it's still the original one, while I think he has changed his. 
I will check on that.

AlexeyAB, 2021-01-13:
Previously logistic activation (sigmoid) was used inside [yolo] layer. 
Currently we use logistic activation (sigmoid) in the previous conv-layer.

mive93, 2021-01-22:
Fixed with the last commit https://github.com/ceccocats/tkDNN/commit/adac8576b0faf515ad3f459b1f50fd16cef6d64d

closed 2021-11-25 by mive93