ceccocats/tkDNN #180: Yolov4-scaled

AleBasso80, 2020-12-22:
Hi,
I tried to export the AlexeyAB yolov4-scaled weights by means of your forked darknet version. I got an assert error at layer 273, I guess that this network is not supported yet. Do you have any plan to do it?

Thanks
Alessandro

erikguo, 2020-12-24:
There's a pull request to support logistic activation. You can tried.    But I found the layer activation just before [yolo] layer is using linear not logistic.

mive93, 2021-01-04:
Hi @AleBasso80 
I did not know about this new addition.
I will look on that in the next days.

AleBasso80, 2021-01-08:
Hi @mive93,
thanks. Its developers claim that this is new state-of-the-art detector.

AlexeyAB, 2021-01-13:
@mive93 
Previously logistic activation (sigmoid) was used inside [yolo] layer.
Currently we use logistic activation (sigmoid) after the previous conv-layer.

LukeAI, 2021-01-21:
@mive93  scaled-yolo is SOTA - would be great to see support for it here

AlexeyAB's medium: https://alexeyab84.medium.com/scaled-yolo-v4-is-the-best-neural-network-for-object-detection-on-ms-coco-dataset-39dfa22fa982

Paper: https://arxiv.org/abs/2011.08036

github: https://github.com/WongKinYiu/ScaledYOLOv4/tree/yolov4-large

mive93, 2021-01-22:
Hi all @AlexeyAB @LukeAI @AleBasso80 @erikguo.
Scaled Yolo 4 is now supported 

https://github.com/ceccocats/tkDNN/commit/adac8576b0faf515ad3f459b1f50fd16cef6d64d

AleBasso80, 2021-01-25:
Hi @mive93, wonderful... Thanks a lot!!

closed 2021-11-25 by mive93