ceccocats/tkDNN #181: cmake: CUDA_cublas_device_LIBRARY variable set to NOTFOUND

kakareko, 2020-12-22:
Hello

CMake fails with an error message:
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_cublas_device_LIBRARY (ADVANCED)
```
All dependencies seem to be available. Could you please help?

Full CMake output:
```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda (found suitable version "10.2", minimum required is "9.0") 
-- Found CUDNN: /usr/lib/aarch64-linux-gnu/libcudnn.so
-- Found CUDNN include: /usr/include
-- Found NVINFER: /usr/lib/aarch64-linux-gnu/libnvinfer.so
-- Found NVINFER include: /usr/include/aarch64-linux-gnu
-- Found CUDNN: /usr/lib/aarch64-linux-gnu/libcudnn.so  
-- Found OpenCV: /usr (found version "4.1.1") 
install dir:/usr/local
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_cublas_device_LIBRARY (ADVANCED)
    linked by target "tkDNN" in directory /home/robot/projects/tkDNN

-- Configuring incomplete, errors occurred!
See also "/home/robot/projects/tkDNN/build/CMakeFiles/CMakeOutput.log".
See also "/home/robot/projects/tkDNN/build/CMakeFiles/CMakeError.log".
```
[CMakeOutput.log](https://github.com/ceccocats/tkDNN/files/5730781/CMakeOutput.log)
[CMakeError.log](https://github.com/ceccocats/tkDNN/files/5730783/CMakeError.log)

AlanNewImage, 2020-12-30:
upgrade your CMake to 3.15 or late.Try 3.18.1

mive93, 2021-01-04:
Hi @kakareko ,

as correctly said by @AlanNewImage, you should update Cmake. As the README says, the project requires the version to be >= 3.15.

Closing for now, feel free to reopen.

closed 2021-01-04 by mive93