ceccocats/tkDNN #184: Runnning on TensorRt 5.0.6.3

tushardhadiwal, 2021-01-05:
Hello,

I have Jetson Nano boards on Jetpack 4.2 in large quantities.
Can I run this repo with TensorRT 5.0.6.3 that comes with Jetpack 4.2?

I see in the readme, it says you were able to run this on Jetson TX2 with Jetpack 4.2.
Which branch should I try? I want to try Yolov4
Thanks

mive93, 2021-01-05:
Hi @tushardhadiwal 
The master should work, have you tried already?

tushardhadiwal, 2021-01-05:
Hi @mive93 
Thanks for quick reply.
I have yolov4.weights file from darknet. I want to run this on a video or rtsp stream. Can you point me on how can I convert that to rt model?

mive93, 2021-01-22:
You have no need to do that, unless you have trained a model on a custom dataset.
Just follow the readme.
When you run the test with the model that you want (e.g. ./test_yolo4) it will download the weights needed from tkDNN and will do everything automatically.

closed 2021-11-25 by mive93