ceccocats/tkDNN #192: AP results confusing

ionicaML, 2021-02-02:
Hello,
I have a question. I have seen on the [YOLOv4 repo](https://github.com/pjreddie/darknet) that the AP50 is 64.9%. However, in the MAP results section it shows that yolov4-tkdnn (416x416) has 0.695, or 69.5% AP50. How is this calculated? I feel like my presumption is wrong, but I can't figure out why. From what I understood, yolov4-tkdnn was only a speedup improvement, so results at AP50 should be around equal, maybe with a 1-2% lower AP.

mive93, 2021-02-08:
Hi @ionicaML 
indeed what you said it's right, tkDNN only optimize yolov4 in terms of execution time, the network and the weights are exactly the same.

In the table that you can find in our README you have the comparison of both the original yolo (darknet column) and our (tkDNN column), both computed via CodaLab (let's say official way) and tkDNN (our implementation of mAP computation).

As you can see original yolov4 AP50 is 0.705 (70,5%), while tkDNN is 0.710 (71,0%), which are almost the same. 
The reason why it is not the same as the one written on the repo/paper is just because probably they were using a greater confidence threshold (e.g. 0.05), while those tests have been performed with a conf threshold of 0.001.

I hope it's clearer now.

ionicaML, 2021-02-08:
Hello,
Thanks for your reply! It's a bit clearer now, but I'm still confused why is there a whole 5% AP difference between CodaLab and the original yolov4 charts? There is quite a difference between 0.7 (from the [README](https://github.com/ceccocats/tkDNN/blob/master/README.md)) and 0.649 (as seen on yolov4 github [here](https://user-images.githubusercontent.com/4096485/101363015-e5c21200-38b1-11eb-986f-b3e516e05977.png)) in terms of AP50 and I'm curious as to where that comes from. Does computation via CodaLab boost performance or are they tested on different data?

closed 2021-02-08 by ionicaML