ceccocats/tkDNN #193: build tensorRT engine on jetson nano get ERROR!

jcyhcs, 2021-02-02:
i download yolo4-csp weights, then i use test_yolo4-csp for convert to TensorRT engine,but got error: look this picture:
[image]
please help!

rickymedrano, 2021-02-07:
Tested it and got same error on the Nano. Not sure if this could be the problem, but yolo4-csp uses logistic activation in a couple layers and it doesn't show logistic as one of the supported activations https://github.com/ceccocats/tkDNN#darknet-parser

mive93, 2021-02-08:
I have just tested that and it works for me. 
Given that I have changed those with my last commit (because the original model changed) be sure to have pulled the latest version, deleted the old weight folder and the old .rt file and create a new one just running ``` ./test_yolo4-csp ``` which automatically downloads the corrects weights and create the correct rt.

rickymedrano, 2021-02-09:
@mive93 I was using the latest version of this repo and was still getting the error.
@MahmoudF98 @jcyhcs 
Figured it out. It was being killed because of insufficient memory. The 2GB swap, using Jetpack 4.5, wasn't enough. You need to increase it to 4GB. Once I set up 4GB swap, with instructions from [here](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap), I was able to create the .rt file successfully for yolo4-csp.

mive93, 2021-02-09:
Oh that's right, I forgot it was on the Nano. 
Then in case you can follow the instructions given here: https://github.com/ceccocats/tkDNN/issues/26.
Indeed, the problem with the Nano is the memory.

closed 2021-11-25 by mive93