ceccocats/tkDNN #196: Cant build new darknet Repo with TkDNN

sharoseali, 2021-02-03:
I re-clone the same darknet repo as mentioned in ReadMe to run CSP models with TkDNN, I change GPU, OpenCV and Cudnn = 1 in Make file (to make it with OpenCV and CUDA) and make but it ends up with the errors like this:
```
`nvcc -gencode arch=compute_72,code=[sm_72,compute_72] -Iinclude/ -I3rdparty/stb/include -DOPENCV `pkg-config --cflags opencv4 2> /dev/null || pkg-config --cflags opencv` -DGPU -I/usr/local/cuda/include/ -DCUDNN --compiler-options "-Wall -Wfatal-errors -Wno-unused-result -Wno-unknown-pragmas -fPIC -Ofast -DOPENCV -DGPU -DCUDNN -I/usr/local/cudnn/include" -c ./src/convolutional_kernels.cu -o obj/convolutional_kernels.o
./src/convolutional_kernels.cu(851): error: expected an expression

./src/convolutional_kernels.cu(862): warning: a user-provided literal suffix must begin with "_"

./src/convolutional_kernels.cu(863): warning: parsing restarts here after previous syntax error

./src/convolutional_kernels.cu(850): warning: variable "old_weights" was declared but never referenced

./src/convolutional_kernels.cu(870): error: expected an expression

./src/convolutional_kernels.cu(877): warning: parsing restarts here after previous syntax error

./src/convolutional_kernels.cu(891): error: identifier "old_weights" is undefined

./src/convolutional_kernels.cu(896): error: expected a declaration

./src/convolutional_kernels.cu(1072): warning: parsing restarts here after previous syntax error

./src/convolutional_kernels.cu(1073): error: identifier "l" is undefined

./src/convolutional_kernels.cu(1078): error: this declaration has no storage class or type specifier

./src/convolutional_kernels.cu(1078): error: expected a ";"

./src/convolutional_kernels.cu(1079): error: this declaration has no storage class or type specifier

./src/convolutional_kernels.cu(1079): error: variable "l" has already been defined

./src/convolutional_kernels.cu(1079): error: expected a ";"

./src/convolutional_kernels.cu(1081): error: expression must have class type

./src/convolutional_kernels.cu(1081): error: expression must have class type

./src/convolutional_kernels.cu(1083): error: this declaration has no storage class or type specifier

./src/convolutional_kernels.cu(1083): error: declaration is incompatible with "void cuda_pull_array(float *, float *, size_t)"
include/darknet.h(1074): here

./src/convolutional_kernels.cu(1083): error: identifier "state" is undefined

./src/convolutional_kernels.cu(1083): error: expected a ")"

./src/convolutional_kernels.cu(1087): error: expected a declaration

./src/convolutional_kernels.cu(1117): warning: parsing restarts here after previous syntax error

./src/convolutional_kernels.cu(1122): error: this declaration has no storage class or type specifier

./src/convolutional_kernels.cu(1122): error: expression must have class type

./src/convolutional_kernels.cu(1122): error: expected a ")"

./src/convolutional_kernels.cu(1129): error: this declaration has no storage class or type specifier

./src/convolutional_kernels.cu(1129): error: identifier "alpha" is undefined

./src/convolutional_kernels.cu(1129): error: expected a ")"

./src/convolutional_kernels.cu(1170): error: expected a declaration

./src/convolutional_kernels.cu(1203): warning: parsing restarts here after previous syntax error

./src/convolutional_kernels.cu(1205): error: expected a declaration

./src/convolutional_kernels.cu(1208): error: this declaration has no storage class or type specifier

./src/convolutional_kernels.cu(1208): error: declaration is incompatible with "void free(void *)"
/usr/local/cuda/include/crt/common_functions.h(150): here

./src/convolutional_kernels.cu(1208): error: a value of type "float *" cannot be used to initialize an entity of type "int"

./src/convolutional_kernels.cu(1209): error: this declaration has no storage class or type specifier

./src/convolutional_kernels.cu(1209): error: variable "free" has already been defined

./src/convolutional_kernels.cu(1209): error: identifier "a_avg" is undefined

./src/convolutional_kernels.cu(1210): error: expected a declaration

./src/convolutional_kernels.cu(1294): warning: parsing restarts here after previous syntax error

./src/convolutional_kernels.cu(1300): error: this declaration has no storage class or type specifier

./src/convolutional_kernels.cu(1300): error: declaration is incompatible with "void reset_nan_and_inf(float *, size_t)"
./src/blas.h(147): here

./src/convolutional_kernels.cu(1300): error: expression must have class type

./src/convolutional_kernels.cu(1300): error: expected a ")"

./src/convolutional_kernels.cu(1301): error: this declaration has no storage class or type specifier

./src/convolutional_kernels.cu(1301): error: declaration is incompatible with "void fix_nan_and_inf(float *, size_t)"
./src/blas.h(146): here

./src/convolutional_kernels.cu(1301): error: expression must have class type

./src/convolutional_kernels.cu(1301): error: expected a ")"

./src/convolutional_kernels.cu(1304): error: expected a declaration

At end of source: warning: parsing restarts here after previous syntax error

41 errors detected in the compilation of "/tmp/tmpxft_00000e3c_00000000-6_convolutional_kernels.cpp1.ii".
Makefile:185: recipe for target 'obj/convolutional_kernels.o' failed
make: *** [obj/convolutional_kernels.o] Error 1`
```
Can you guide, where is the issue...... @ceccocats  @mive93 
Thanks in advance

rickymedrano, 2021-02-07:
On main page 

> Compile with make without modify anything in the makefile, it will use only CPU

So use original makefile without any changes. There is no need to compile with CUDA and cuDNN as you're only using that fork of darknet to convert the weights to an appropriate format for tkDNN.

mive93, 2021-02-08:
Hi @sharoseali,

everything @rickymedrano said is correct (thanks :) ). 
For tkDNN purposes you only need to compile it with CPU and nothing else.

sharoseali, 2021-02-08:
> Hi @sharoseali,
> 
> everything @rickymedrano said is correct (thanks :) ).
> For tkDNN purposes you only need to compile it with CPU and nothing else.

Thanks @rickymedrano @mive93 . Yes it was the issue. Now it worked.. Closing the issue

closed 2021-02-08 by sharoseali