ceccocats/tkDNN #197: execution error for previously created trt model

syjeon121, 2021-02-04:
hi, i recently updated the tkdnn code.
however the previously created yolov4 trt model does not run after the update.
```
RtBuffer 0   dim: Data dim: 1 3 608 608 1
RtBuffer 1   dim: Data dim: 1 33 76 76 1
RtBuffer 2   dim: Data dim: 1 33 38 38 1
RtBuffer 3   dim: Data dim: 1 33 19 19 1
camera started
[1]    4584 floating point exception (core dumped)  ./demo  ../demo/yolo_test.mp4
```

i remember getting my old code in Jun of last year
I think it's because the code inside the layer has changed. Is that right??

mive93, 2021-02-08:
Hi @syjeon121, 

yes exactly. You need to re-create the rt file and it will work again!

syjeon121, 2021-02-08:
@mive93 thanks!

closed 2021-02-08 by syjeon121