ceccocats/tkDNN #198: Issue when trying to download fork for darknet weight export

pladosz, 2021-02-05:
Hi,

When I try to download repository necessary to export weights from darknet [https://git.hipert.unimore.it/fgatti/darknet.git], I get error:

> fatal: unable to access 'https://git.hipert.unimore.it/fgatti/darknet.git/': Failed to connect to git.hipert.unimore.it port 443: Operation timed out

Could you please advise?

Thanks!

rickymedrano, 2021-02-07:
I just tested it and this works
`git clone https://git.hipert.unimore.it/fgatti/darknet.git`

pladosz, 2021-02-08:
Hi,

I just checked with internet tethered from my phone and indeed it works. Could it be that my university network blocks this specific port?

rickymedrano, 2021-02-08:
> Hi,
> 
> I just checked with internet tethered from my phone and indeed it works. Could it be that my university network blocks this specific port?

No idea, that's on your end and you would need to investigate it. It could be your university network blocking certain sites or the hipert.unimore.it site that was temporarily down.

mive93, 2021-02-08:
Hi @pladosz, 

yeah, we had problems with the university gitlab. Now it should be fixed.
Thanks @rickymedrano for the quick suppport.

pladosz, 2021-02-08:
Thanks for all the help and replies. I think there is an issue with network settings on our end, as I still can't access it from my university network. I think this can be closed now?

closed 2021-02-16 by pladosz