ceccocats/tkDNN #202: Is there a way similar to trtexec --loadEngine for just testing the speed of the model?

ysyyork, 2021-02-25:
For pure tensorrt engines, I can easily test the throughput by `trtexec --loadEngine` and it will randomly generate the inputs and measure the inference time. Is there a similar function here? Thanks.

rickymedrano, 2021-03-11:
+1, would also like to be able to test theoretical inference. Tried using the .rt file with trtexec but it doesn't work.

mive93, 2021-03-27:
Hi @ysyyork  and @rickymedrano , you can use the test_rtinference, e.g., with yolov4 and 1 batch:
```
    ./test_rtinference yolo4_fp32.rt 1 
```

ysyyork, 2021-03-28:
> Hi @ysyyork and @rickymedrano , you can use the test_rtinference, e.g., with yolov4 and 1 batch:
> 
> ```
>     ./test_rtinference yolo4_fp32.rt 1 
> ```

Thanks @mive93 ! this is very helpful!

closed 2021-07-02 by mive93