ceccocats/tkDNN #203: Could not parse line

lijunnankman, 2021-03-02:
When I use "./test_yolo4", it came that "could not parse line:"

lijunnankman, 2021-03-09:
@ceccocats @mive93   
[image]
just like this ，could you help me？

mive93, 2021-03-27:
Hi @lijunnankman, 
is it from the master branch? Or are you using a custom yolov4?

lijunnankman, 2021-03-30:
Hi，thanks for your reply.When I trained yolov4, I changed the anchors in the cfg file. Is this influential, please?

mive93, 2021-04-02:
No, it's not a problem because it does not read the anchors from there. 
Can you do a git status to check for any other changes. Or can you upload the file you're using so I can check?

closed 2021-11-25 by mive93