ceccocats/tkDNN #205: DIMENSIONS WHILE CREATING RT FILE

chandu1263, 2021-03-10:
How to change the dimensions while creating .rt file?
One way I am using currently is by changing the height and width parameters in cfg while exporting layers from weights and cfg file, but I am not sure if it's correct way. Is there any environment variable like batchsize, FP16 mode ?
Thanks in Advance!

chandu1263, 2021-03-10:
@mive93 could you help me with this?

Sudhakar17, 2021-03-24:
@chandu1263 and @mive93 : Did you find the solution?

mive93, 2021-03-27:
Hi @chandu1263  and @Sudhakar17, if you want to change the input dimension of the network you have to export again the weights form darknet. 
Follow this guide: https://github.com/ceccocats/tkDNN#1export-weights-from-darknet. 
Just be sure to put the wanted dimension in the cfg you use for the export.

closed 2021-11-25 by mive93