ceccocats/tkDNN #207: Bounding Box too big

AGpix, 2021-03-18:
Hello I am using Yolov4-tiny weights exported from darknet but this results in bounding boxes which are double the size of the object. Any suggestion why this could be

AGpix, 2021-03-18:
Using coco yolov4-tiny.weights
[image]

mive93, 2021-03-27:
Hi @AGpix , 
which configuration file are you using? Have you modified the test?
Have you tried yolov4tiny that is already in this repo?

AGpix, 2021-04-06:
> Hi @AGpix ,
> which configuration file are you using? Have you modified the test?
> Have you tried yolov4tiny that is already in this repo?
Hello, we solved there was an error on our part in the cfg file. Thanks

closed 2021-11-25 by mive93