ceccocats/tkDNN #210: [dependencies] Compilation Errors with NVIDIA Jetson TX2

ZexinLi0w0, 2021-03-20:
The error log:
[  1%] Linking CXX shared library libkernels.so
[ 17%] Built target kernels
Consolidate compiler generated dependencies of target tkDNN
[ 18%] Linking CXX shared library libtkDNN.so
[ 44%] Built target tkDNN
Consolidate compiler generated dependencies of target test_yolo3_512
[ 45%] Linking CXX executable test_yolo3_512
/root/miniforge3/lib/libopencv_dnn.so.4.5.1: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'
/root/miniforge3/lib/libopencv_dnn.so.4.5.1: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()@GLIBCXX_3.4.26'
collect2: error: ld returned 1 exit status
CMakeFiles/test_yolo3_512.dir/build.make:157: recipe for target 'test_yolo3_512' failed
make[2]: *** [test_yolo3_512] Error 1
CMakeFiles/Makefile2:151: recipe for target 'CMakeFiles/test_yolo3_512.dir/all' failed
make[1]: *** [CMakeFiles/test_yolo3_512.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2

================================================
It seems to be a GLIBCXX_3.4.26 missing, and I check it by:
```
strings /usr/lib/aarch64-linux-gnu/libstdc++.so.6 | grep GLIBCXX
```
and didn't find GLIBCXX_3.4.26. (the highest version in my system is GLIBCXX_3.4.25)
```
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_3.4.22
GLIBCXX_3.4.23
GLIBCXX_3.4.24
GLIBCXX_3.4.25
GLIBC_2.17
GLIBC_2.18
GLIBCXX_DEBUG_MESSAGE_LENGTH
```

I have tried sudo apt update && sudo apt upgrade, the GLIBCXX should be the newest version. What can I try to fix this? Does it mean that I shoudl compile a new version gcc/g++?

My environment is shown as below:
NVIDIA Jetson TX2
System: Ubuntu 18.04.5 LTS
CUDA 10.2.89
CUDNN 8.0.0.180
libopencv 4.1.1
gcc 7.5.0
g++ 7.5.0

rickymedrano, 2021-03-22:
I ran your grep command on my Jetson Nano and also didn't see any GLIBCXX 3.4.26. Wanted to let you know as I'm able to compile tkDNN successfully so I'm not sure that is the issue. Do you have TensorRT installed?

ZexinLi0w0, 2021-03-23:
> I ran your grep command on my Jetson Nano and also didn't see any GLIBCXX 3.4.26. Wanted to let you know as I'm able to compile tkDNN successfully so I'm not sure that is the issue. Do you have TensorRT installed?

Yes, I install the dependencies by running
```
sudo apt install nvidia-jetpack
```

```
dpkg -l |grep tensorrt
==========
ii  nvidia-container-csv-tensorrt              7.1.3.0-1+cuda10.2                               arm64        Jetpack TensorRT CSV file
ii  nvidia-tensorrt                            4.4.1-b50                                        arm64        NVIDIA TensorRT Meta Package
ii  tensorrt                                   7.1.3.0-1+cuda10.2                               arm64        Meta package of TensorRT
```

rickymedrano, 2021-03-23:
I ran dpkg -l | grep tensorrt as well and have the same as you except for:
`nvidia-tensorrt 4.4.1-b50`
I wonder if that package is giving you issues? Could be worthwhile removing and then try building tkDNN again.

ZexinLi0w0, 2021-03-23:
> I ran dpkg -l | grep tensorrt as well and have the same as you except for:
> `nvidia-tensorrt 4.4.1-b50`
> I wonder if that package is giving you issues? Could be worthwhile removing and then try building tkDNN again.

After removing nvidia-tensorrt, I still meet the same compiling issues.

ZexinLi0w0, 2021-03-23:
The cmake log is
```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found CUDA: /usr/local/cuda (found suitable version "10.2", minimum required is "9.0")
-- Found CUDNN: /usr/lib/aarch64-linux-gnu/libcudnn.so
-- Found CUDNN include: /usr/include
-- Found NVINFER: /usr/lib/aarch64-linux-gnu/libnvinfer.so
-- Found NVINFER include: /usr/include/aarch64-linux-gnu
-- Found CUDNN: /usr/lib/aarch64-linux-gnu/libcudnn.so
-- Found OpenCV: /root/miniforge3 (found version "4.5.1")
install dir:/usr/local
-- Configuring done
-- Generating done
```

rickymedrano, 2021-03-23:
I think it might be related to your installation of OpenCV. The error happens at
`/root/miniforge3/lib/libopencv_dnn.so.4.5.1`
which for me, my OpenCV is installed in /usr/lib/aarch64-linux-gnu/libopencv_dnn.so.4.1.1 which is also an older version than yours. I'm not sure why yours is installed in the root folder, I dont think it should be there. Have you installed OpenCV with apt?

ZexinLi0w0, 2021-03-23:
> I think it might be related to your installation of OpenCV. The error happens at
> `/root/miniforge3/lib/libopencv_dnn.so.4.5.1`
> which for me, my OpenCV library is installed in /usr/lib/aarch64-linux-gnu/libopencv_dnn.so.4.1.1 which is also an older version than yours. I'm not sure why yours is installed in the root folder, I dont think it should be there. Have you installed OpenCV with apt? Your cmake log shows your OpenCV libary is:
> `- Found OpenCV: /root/miniforge3 (found version "4.5.1")`

I build an opencv 4.5.1 from source earlier, also I have /usr/lib/aarch64-linux-gnu/libopencv_dnn.so.4.1.1.

rickymedrano, 2021-03-23:
You need to get Cmake to point to that 4.1.1 version of libopencv_dnn because it currently points to your /root/minieforge/lib directory. I'm not very experiencing with cmake though so I'm not sure how you would do that.

ZexinLi0w0, 2021-03-23:
> I think CMake is having issues then with pointing to the correct libopencv_dnn because it currently points to your /root/minieforge/lib directory whereas mine is in /usr/lib/aarch64-linux-gnu/. I'm not very experiencing with cmake though so I'm not sure how you would fix that.

After I modify the CMakeLists.txt, I can build the project correctly. Thank you very much.
```
#find_package(OpenCV REQUIRED)
#set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -DOPENCV")

include_directories("/usr/include/opencv4/")
link_directories("/usr/lib/aarch64-linux-gnu/")
link_libraries(
        opencv_calib3d
        opencv_core
        opencv_dnn
        opencv_features2d
        opencv_flann
        opencv_gapi
        opencv_highgui
        opencv_imgcodecs
        opencv_imgproc
        opencv_ml
        opencv_objdetect
        opencv_photo
        opencv_stitching
        opencv_videoio
        opencv_video)
```

closed 2021-03-23 by ZexinLi0w0