ceccocats/tkDNN #214: Support for cuDNN 8.1.1

syjeon121, 2021-04-01:
hi @mive93 
according to cudnn documentation, the performance of RTX 3090 has improved in 8.1.1
so i am currently testing on cuDNN 8.1.1, TensorRT 7.2.3.4 using RTX 3090 

can i see these improvement in current tkDNN code?

mive93, 2021-04-02:
I don't get the question. 
Are you asking if CuDNN 8.1.1 is supported?

syjeon121, 2021-04-02:
thanks @mive93 
yes, and TensorRT 7.2.3.4 too

perseusdg, 2021-04-02:
@syjeon121 I was able to use tkdnn with cudnn 8.1.1 and tensorrt 7.2.3 on ubuntu 20.04

mive93, 2021-04-02:
Thanks @perseusdg :)

syjeon121, 2021-04-06:
@mive93 @perseusdg  hi, sorry to reopen this issue
one thing i'm curious about is how to use internal function according to the cuDNN version up
for example convolution forward function

or is it automatically applied by simply upgrading the version?

perseusdg, 2021-04-06:
@syjeon121 are you asking how tkdnn is able to work with different versions of cudnn ?

syjeon121, 2021-04-06:
@perseusdg yes. so i want to apply cuDNN 8.1.1 function if there is something new

perseusdg, 2021-04-06:
you can check [this commit](https://github.com/ceccocats/tkDNN/commit/c4aad7fe95e0f8ed45fd175d565d1926d2880f6c) out to see the changes made to make the library work with cudnn 8.x and going by the improvements/new features mentioned in the release notes of cudnn 8.1.1 i dont think there is anything new that is pertinent to this library

syjeon121, 2021-04-07:
@perseusdg ok, i will check the commit
thanks!

closed 2021-11-25 by mive93