ceccocats/tkDNN #215: Keep getting "Not supported field" error

Duarte-Nunes, 2021-04-06:
High, im running the repository on the Jetson Nano 2Gb, and i keep getting this error (sorry for the bad quality, had to take the picture with my phone), i tried rebooting but it didnt change.

[image]

mive93, 2021-04-07:
Hi @Duarte-Nunes, check this: https://github.com/ceccocats/tkDNN/issues/26

Duarte-Nunes, 2021-04-13:
Some further elaboration, i do believe that the problem is still low memory however i am running the jetson only on terminal the max memory usage is 500mb which leaves around 6 gb of free memory (including swap) but when i run the test it says "GPU free memory: 100mb" or 70 mb ou around those values

Duarte-Nunes, 2021-04-13:
@mive93 Nevermind it appears through some research on Nvidia forums that Cuda device memory cannot be swaped and therefore even with the trick posted on the Issue you linked my Jetson nano which is a 2Gb model cannot run the code.

closed 2021-04-20 by Duarte-Nunes